Duet3D/DuetSoftwareFramework #1: Interruptible gcode commands

techildllc, 2019-07-01:
M280 command seems to be able to interrupt/ occur in parallel with G1 move commands. 

For example:
*Current X location is at 0*
G1 X300 F5000
M280 P0 S175

While moving to the 300 in the X, the servo will actuate instead of actuating after arriving at X300.

Currently must use M400 after the G1 command to gate the M280 command.

chrishamm, 2019-07-01:
It really depends on where you execute the macro file and/or M280. Depending on the code channel, certain codes including M280 are queued for just-in-time execution. If the macro is executed via the HTTP channel (web interface), the code queue is not used which causes M280 to be executed when it is interpreted and not when the move has been finished.

This behaviour is primarily caused by RepRapFirmware and probably not related to DSF itself.

techildllc, 2019-07-01:
Is this the code queue suppose to be respected if this was executed within a user or sys macro? If so, then it’s not behaving as intended.

chrishamm, 2019-07-01:
It does not matter if it is a user or sys macro, it entirely depends on where you execute the macro/code. The code queue is only used on the file print, serial (USB) and Telnet channels AFAIR.

techildllc, 2019-07-01:
Ok so this is an intended effect on RRF3? As this was not how RRF2 worked.

chrishamm, 2019-07-01:
In this regard there should be no difference between RRFv2 and RRFv3 - the corresponding code was barely modified after RRFv2.

Where do you attempt to execute M280, via the web interface or via a file print?

techildllc, 2019-07-01:
This was executed from the web interface. More specifically, this was part of the tool change script (ie tpre, tpost, tfree). So for example, in tfree, there is a G1 command that moves the carriage to a specific location, then calls for a M280 to unlock the tool head. Well what happens is while the carriage is still on the move before it arrives at the dock, the tool head is unlocked and dropped.

This did not happen in RRF2.

dc42, 2019-07-01:
The behaviour needs to be changed so that if the command received over the SBC interface was fetched from a file (whether a macro file or a regular print file), the M280 is queued for just-in-time execution. I think it makes sense to do this for all channels when the command is within a macro file. I will implement this change later this week.

techildllc, 2019-07-01:
Sorry, a little new at the terminology here, can you clarify what you mean by "just-in-time execution"? I assume this means the M280 will be queued and will wait for previous commands to complete?

chrishamm, 2019-07-02:
Yes that is correct.

@dc42 in my RRFv3 branch I added a new function to the GCodeBuffers called IsDoingFile which is true if either a print or macro file is being performed (works both for Duet2+DSF). We could use that for the buffering detection.

dc42, 2019-07-02:
I have used IsDoingFile to fix this in RRF 2019-07-02b2. The fix works for the case that the macro file is commanded by the Duet, for example by sending a M98 command from USB. It does not work if the M98 command is sent from the GCode Console of DWC because the fileId member of GCodeMachineState doesn't get set. Does fixing that require a change to DSF?

dc42, 2019-07-02:
New RRF binary available at https://www.dropbox.com/sh/3azy1njy3ayjsbp/AACquxr2m00eV568RZg5QG5wa?dl=0

techildllc, 2019-07-03:
I flashed the new binary, when launching DCS, I get to "connecting to reprapfirmware...." and I get "Error: Duet is not available". Is DCS looking for a particular version? I am pretty sure I have it flashed correctly, did the same thing as I did the first time around.

chrishamm, 2019-07-05:
I've changed the latest RRF version to queue codes whenever a file is being executed. If it does not work, feel free to reopen this issue.

closed 2019-07-05 by chrishamm