Duet3D/DuetSoftwareFramework #2: sudo systemctl enable fails on stretch

beikeland, 2019-07-01:
Followed https://duet3d.dozuki.com/Wiki/Duet_3_prototype_guide_for_OEMs

`sudo systemctl start duetcontrolserver` works without errors but enable fails as shown below.
same problem with duetwebserver

```
pi@octopi:~ $ sudo ln -s /opt/dsf/services/duetcontrolserver.service /etc/systemd/system/duetcontrolserver.service
pi@octopi:~ $ sudo ln -s /opt/dsf/services/duetwebserver.service /etc/systemd/system/duetwebserver.service
pi@octopi:~ $ sudo systemctl daemon-reload
pi@octopi:~ $ sudo systemctl start duetcontrolserver
pi@octopi:~ $ sudo systemctl |grep  duetcontrolserver
duetcontrolserver.service                                                                            loaded active running   Duet Control Server                                      
pi@octopi:~ $ sudo systemctl enable duetcontrolserver
Failed to enable unit: File duetcontrolserver.service: No such file or directory
```

and stack trace if that helps
[strace.log](https://github.com/chrishamm/DuetSoftwareFramework/files/3347082/strace.log)

beikeland, 2019-07-01:
more digging shows /opt/dsf/services/ didn't exist, nor the service files below.

quick fix
```
sudo mkdir -p /opt/dsf/services/
sudo wget https://github.com/chrishamm/DuetSoftwareFramework/raw/master/pkg/duetcontrolserver/usr/lib/systemd/system/duetcontrolserver.service -O /opt/dsf/services/duetcontrolserver.service
sudo wget https://github.com/chrishamm/DuetSoftwareFramework/raw/master/pkg/duetwebserver/usr/lib/systemd/system/duetwebserver.service  -O /opt/dsf/services/duetwebserver.service
```

chrishamm, 2019-07-01:
Thanks for the note - I just updated the instructions. Btw, you don't need to symlink the service files any more, they are now part of the systemd service directory (hence the path to /usr/systemd/...).

closed 2019-07-01 by chrishamm