Duet3D/DuetSoftwareFramework #5: Nested user macro call inside sys macro error

techildllc, 2019-07-01:
Currently unable to run a user macro nested inside a sys macro. System hangs when it gets to that step. Although regular macro calling another regular macro works as intended.

chrishamm, 2019-07-01:
I am still in the process of implementing buffered G-codes for higher bandwidth which will bring a lot of more changes/fixes. I'll get back to this when I have that part working stable so you could test the current scenario again.

I already found one potential race condition that will be addressed in the next release.

techildllc, 2019-07-01:
ok, perfect, just logging these issues so others are aware and we can remember to go back and revisit them. 

Current work around is just to manually insert all the nested macro gcodes in the sys macro.

chrishamm, 2019-07-05:
Please test again with DCS v1.0.1.0 and the latest firmware version - see https://duet3d.dozuki.com/Wiki/Duet_3_prototype_guide_for_OEMs?revisionid=HEAD#Section_Updating_DSF

chrishamm, 2019-07-21:
Closing this issue - if the problem persists, feel free to reopen.

closed 2019-07-21 by chrishamm