Duet3D/DuetSoftwareFramework #12: Filenames are case-sensitive

dc42, 2019-07-02:
Filenames in RRF have always been case-insensitive in the past, however in DSF they are case-sensitive. For example, if a M98 command is used to run a macro and the filename is not in exactly the correct case, then the file is not found.

chrishamm, 2019-07-05:
I don't think this should be addressed in DSF, it's an OS issue in the first place because it decides how to file paths are resolved. AFAIR on NTFS and FAT volumes, filenames on Linux are already treated case-insensitive and I heard ext4 has just introduced another config flag to turn off case-sensivity, too.

So I believe we should leave it up to the user to decide if a file names are supposed to be handled case-sensitive or insensitive.

beikeland, 2019-07-05:
With the risk of sticking my nose where it doesn't belong ... I'd tend to agree that one shouldn't try to force case insensitivity if the underlying operating/file-system supports case sensitivity as it would be possible to have two files A, and a - and no way to know which is which if DSF is to try to deal with it.

An acceptable workaround could be a FUSE or loopback FAT filesystem for macros if its a concern to break existing configs.

dc42, 2019-07-06:
I agree that it's an OS issue; but we must at least document this change, and preferably provide instruction on how to switch the file system to be case insensitive if that can be done easily. Most existing programs generate GCode commands using uppercase characters only, which may cause problems if users try to connect such programs to DSF.

beikeland, 2019-07-06:
Given the concept of the "virtual sd card" already, its probably the easiest to take that one step further and turn it into a an actualy case insensitive filesystem. Relying on code in DSF to handle the case insensitivity could lead to problems if the underlying filesystem makes different assumptions based on locale or whatnot.

This looks easy enough to do http://www.brain-dump.org/projects/ciopfs/ other otions include making a partition on the Pi's card, a disk image + loopback mount, USB stick and probably more.

chrishamm, 2019-08-07:
I am closing this because it is more a hardware configuration issue than a problem with DSF itself.

In the longterm I advise using Linux 5.2 and disabling case-sensitivity via an ext4 mount option. See https://www.phoronix.com/scan.php?page=news_item&px=EXT4-Case-Insensitive-Sent

closed 2019-08-07 by chrishamm