Duet3D/DuetSoftwareFramework #15: wiringpi - missing dependancy

beikeland, 2019-07-03:
Fresh install on debian strech (lite):

```pi@raspberrypi:~ $ sudo /opt/dsf/bin/DuetControlServer
Duet Control Server v1.0.0.14
Written by Christian Hammacher for Duet3D
Licensed under the terms of the GNU Public License Version 3

Loading settings... Done!
Initialising object model... Done!
Connecting to RepRapFirmware... wiringPiISR: Can't find gpio program
```
solved by `sudo apt-get install wiringpi `

techildllc, 2019-07-03:
I didn't have this problem with "Raspbian Stretch with desktop and recommended software". Must be something that was not included in the lite version.

beikeland, 2019-07-03:
> I didn't have this problem with "Raspbian Stretch with desktop and recommended software". Must be something that was not included in the lite version.

Yeah, most likely, but it should still be added to the dependency list for the duetcontrolserver (or duetruntime?). Couldn't quite make sense of the build.sh script for the package as it fetches files from a local mount and git.

> By the way, are you running the latest RRF binary? "Duet3Firmware-2012-07-02b2.bin"

yeah, just got the thing today so that was the latest version I could find.

techildllc, 2019-07-03:
> yeah, just got the thing today so that was the latest version I could find.

As in you were able to flash your board using BOSSA to that version of the bin? I could not get my board to work with the PI after the upgrade. When launching DCS, I get to "connecting to reprapfirmware...." and I get "Error: Duet is not available". Dropping back to previous build works fine.

beikeland, 2019-07-03:
I _think_ so; web interface reports RepRapFirmware for Duet 3 v0.5 version 3.0b6 running on Duet 3 prototype v0.5 and when launching DCS:
```
$ sudo /opt/dsf/bin/DuetControlServer
Duet Control Server v1.0.0.14
Written by Christian Hammacher for Duet3D
Licensed under the terms of the GNU Public License Version 3

Loading settings... Done!
Initialising object model... Done!
Connecting to RepRapFirmware... [warn] RepRapFirmware got a bad header checksum
Done!
Creating IPC socket... Done!
```
It did spit out a warning, but seems to work, havent connected the other hardware yet.

I did notice the Pi would not connect to the Duet if you power the Pi from USB, and the Duet gets 5v off the Pi.  Havent tried with 24v supply or the 5v_in jumper as USB power to the Duet and 5v from the Duet to the Pi works.

techildllc, 2019-07-03:
Bad header checksum is a "known" issue. Right, you need a 12 or 24v supply to also power your PI. I have my printer up and running since "3.0b3-ch".

Interesting, must be something odd that occurred with my flashing. The new version looks to be 3.0b6. I will try again tonight.

dc42, 2019-07-03:
To check the firmware version, send M115 from GCode Console. The firmware date should be 2012-07-02b2.

beikeland, 2019-07-03:
Confirmed `FIRMWARE_DATE: 2019-07-02b2`- just didn't recall the code and figured the web interface would have the same info.

techildllc, 2019-07-03:
Thanks for that info, I will reflash and try again tonight.

dc42, 2019-07-03:
The web interface doesn't currently display the firmware date.

chrishamm, 2019-07-05:
Thanks, I've added wiringpi to the list of DCS dependencies. Should be fixed in the latest 1.0.1.0 version.

closed 2019-07-05 by chrishamm