Duet3D/DuetSoftwareFramework #18: Shown Fan RPM increases as speed decrease

beikeland, 2019-07-07:
Possibly a RRF 3 issue and not a DFS issue?

In any case after configuring fan with tach output the shown RPM increases as the fan is slowed down.

3 pin fan 
positive --> V_OUTLC1
negative --> OUT_4
tach --> OUT4.tach

```
M950 F2 C"out4+out4.tach" ;Fan RPM not shown 
M106 P2 S0    ;Fan RPM not shown 
M106 P2 S1    ;Fan RPM show ~5500 RPM
M106 P2 S0.5  ;Fan RPM show ~9000 RPM
```

dc42, 2019-07-08:
When using 4-wire PWM fans you need to invert the output to get the speed controlled in the correct sense:

M950 F2 C"!out4+out4.tach"

beikeland, 2019-07-08:
The only 4 pin fan i had to try didn't seem to give a rpm signal at all, so I went with 3 pin fans and removed the inversion. Didn't seem to make any difference with or without ^ to enable the internal pull up either. But I'll try to get some measurements done to see whats what

beikeland, 2019-07-08:
Scope probes were hiding, but I scavenged a 12v fan from an old computer that was tested before removal.

When connected with the original 3 pin plug, to GND, V_OUTCL1 and OUT4.tach it runs fixed speed with the correct RPM.

When connected to V_OUTCL1, OUT4.tach and OUT_4  it runs at the correct RPM when running at full speed, but not when throttled. I'm leaning towards this being a limitation of the fans controller not being able to output reliable RPM signal if fed a chopped supply. But will confirm when I find where the scope probes are hiding.
```
S   Tach Duet
1.0 3400  3400
0.9 3200 10000
0.8 3000  9000
0.7 2800  8800
0.6 2500 10000
0.5 2200  8000
0.4 1700  7500
0.3 1500  7500
0.2  900  6700
0.1  200  6600
```
S= S parameter from M106. Tach = Measured with tachometer. Duet = reported by Duet.

dc42, 2019-07-09:
The tacho does not work on 3-pin fans when you PWM the power to them.

closed 2019-07-09 by dc42