Duet3D/DuetSoftwareFramework #22: Memory leak in DCS

wilriker, 2019-07-17:
`DuetControlServer` has a memory leak. It will request additional ~80MiB/hour. This is linear and does not depend on any process being connected as a client to `DCS`.

chrishamm, 2019-07-21:
I suspect this is related to anonymous JSON deserialization in the DCS status update thread and a problem caused by the JSON.NET library. Can you check if the problem persists with the latest DSF version?

That particular code is still temporary and will be replaced soon, but I already added some further improvements to that part of the latest DCS version.

wilriker, 2019-07-21:
I'll tell you tomorrow. ;-)

wilriker, 2019-07-22:
It seems that whatever you did made it worse. The additional memory consumption was increased to 90-100MiB/hour but does not seem to be linear anymore (according to my few datapoints so far). I will collect some more data throughout the next days.

dc42, 2019-07-22:
Could it be simply that DSF hasn't been given a memory limit, so as the memory is there it's grabbing it in preference to doing garbage collection? I presume there is a way of setting a memory limit.

wilriker, 2019-07-22:
I don't know .NET Core enough to say how memory is limited. A quick search also did not reveal anything. I know from Java that if you do not specify anything it will be limited to 25% of total RAM but obviously that's not the case for .NET Core. I will try to find a setting if possible and check with a very low value if it
a) garbage collects and continuous or
b) dies with OutOfMemoryError

wilriker, 2019-07-29:
This is fixed in 1.0.2.2.

closed 2019-07-29 by wilriker