Duet3D/DuetSoftwareFramework #25: Heaters shut off while printing

wilriker, 2019-07-17:
While printing all heaters will be shut-off without any further notice after a couple of minutes. Trying to re-enable them via DWC leads to the print failing because of #24.

chrishamm, 2019-07-21:
Do you get any particular message when this happens? I haven't encountered this problem before.

wilriker, 2019-07-21:
No, unfortunately no message regarding heaters at all. Also nothing heater related in `DCS` log output. I will retry with 1.0.2.1 probably end of next week to see if I can provide some more information.

chrishamm, 2019-08-07:
I still cannot reproduce this even though I ran various test prints today. I suggest you enable debugging for GCodes, Heat and perhaps LinuxInterface via M111 and check if you can discover anything new.

wilriker, 2019-08-09:
Since I reported this there was no combination of me having the time to print something and having a workable firmware/DSF combination at the same time. I will probably retest this today.

wilriker, 2019-08-09:
Just completed the first successful print. I close this and reopen if it ever happens again.

closed 2019-08-09 by wilriker