Duet3D/DuetSoftwareFramework #27: DWC not continuously updating anymore

wilriker, 2019-07-21:
`DCS`: 1.0.2.1
`DWS`: 1.0.1.1
`DWC`: 2.0.0-RC7
`RRF`: `1c85904a`

Since updating to the latest DuetSoftwareFramework and RRF `DWC` no longer continuously updates values like temperatures, current VIN value, DHT sensor values. Sometimes it also does not populate the tools section of DWC. This is with the printer being idle. I did not yet test while printing.

beikeland, 2019-07-21:
Confirmed, hovering over the Vin value does show a tool tip with min/max so there is at least three datapoints.
[image]

wilriker, 2019-07-23:
Temperatures are not updated while heating (when temp set manually via DWC).

wilriker, 2019-07-29:
Fixed in `DCS` 1.0.2.2.

wilriker, 2019-08-05:
Reopening since this is again the case with `DCS` 1.0.3.0.

wilriker, 2019-08-05:
Now here is something really strange: If `DCS` gets killed by an error like described in #34 then after it runs again the updates are working and the WebSocket will see continuous updates. This also works across page reloads.

Restarting `DCS` with a working `config.g` does not allow it to work. 

I tracked it down to
```
; Tools
M563 P0 S"Hotend" D0 H1                            ; Define tool 0
```

If that's in `config.g` then no updates will flow. If I comment it out status updates work. So most likely this is tied to #47.

chrishamm, 2019-08-07:
Fixed in DCS 1.0.3.1.

closed 2019-08-07 by chrishamm