Duet3D/DuetSoftwareFramework #28: DCS blocks after executing a macro

wilriker, 2019-07-21:
`DCS`: 1.0.2.1
`DWS`: 1.0.1.1
`DWC`: 2.0.0-RC8-b1
`RRF`: `1c85904a`

When clicking a macro in DWC this will be executed but then `DCS` is blocked (on this channel?). In logging it also states it finished the macro but no more commands are accepted. Restarting DCS fixes the problem.

chrishamm, 2019-07-22:
Please share the macro file that leads to this problem. Running G28, which invokes homeall.g, works without problems on my setup.

wilriker, 2019-07-22:
It's simply a macro called `Motors_Off` that contains
```
M84
```
and nothing else.

wilriker, 2019-07-23:
I just found out that the Emergency Stop button is still working even in this blocked state.

wilriker, 2019-07-30:
This is fixed in `DCS` 1.0.2.2.

closed 2019-07-30 by wilriker