Duet3D/DuetSoftwareFramework #29: Unable to start DWS via systemd

wilriker, 2019-07-21:
`DWS`: 1.0.0.3+

I cannot start DWS as systemd unit. The binary will start and has a PID but the webserver itself will not initialize. There is no output to the journal or a manually added output file for logging. It runs fine when started manually via command line. Then it also can be backgrounded with output redirect.

This is running Arch Linux ARM. Does it work on Raspbian?

beikeland, 2019-07-21:
Just updated on Rasbian lite and it seems to work
```
pi@duet3:~ $ apt list 2>/dev/null | grep duet | awk -F'[/ ]' '{print $1" "$3 }'
duetcontrolserver 1.0.2.1
duetruntime 1.0.2.1
duetsd 1.0.1
duetsoftwareframework 1.0.2.1
duettools 1.0.2.1
duetwebcontrol 2.0.0-2
duetwebserver 1.0.1.1

pi@duet3:~ $ sudo systemctl start duetcontrolserver
pi@duet3:~ $ sudo systemctl start duetwebserver
pi@duet3:~ $ systemctl status  duetwebserver
● duetwebserver.service - Duet Web Server
   Loaded: loaded (/usr/lib/systemd/system/duetwebserver.service; disabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-07-21 23:01:01 BST; 14min ago
 Main PID: 1562 (DuetWebServer)
    Tasks: 19 (limit: 4915)
   CGroup: /system.slice/duetwebserver.service
           └─1562 /opt/dsf/bin/DuetWebServer

Jul 21 23:01:31 duet3 DuetWebServer[1562]: info: Microsoft.AspNetCore.Routing.EndpointMiddleware[1]
Jul 21 23:01:31 duet3 DuetWebServer[1562]:       Executed endpoint 'DuetWebServer.Controllers.MachineController.GetFileList (DuetWebServer)'
Jul 21 23:01:31 duet3 DuetWebServer[1562]: info: Microsoft.AspNetCore.Mvc.Internal.ControllerActionInvoker[2]
Jul 21 23:01:31 duet3 DuetWebServer[1562]:       Executed action DuetWebServer.Controllers.MachineController.DownloadFile (DuetWebServer) in 753.8843ms
Jul 21 23:01:31 duet3 DuetWebServer[1562]: info: Microsoft.AspNetCore.Routing.EndpointMiddleware[1]
Jul 21 23:01:31 duet3 DuetWebServer[1562]:       Executed endpoint 'DuetWebServer.Controllers.MachineController.DownloadFile (DuetWebServer)'
Jul 21 23:01:31 duet3 DuetWebServer[1562]: info: Microsoft.AspNetCore.Hosting.Internal.WebHost[2]
Jul 21 23:01:31 duet3 DuetWebServer[1562]:       Request finished in 981.4789ms 200 application/octet-stream
Jul 21 23:01:31 duet3 DuetWebServer[1562]: info: Microsoft.AspNetCore.Hosting.Internal.WebHost[2]
Jul 21 23:01:31 duet3 DuetWebServer[1562]:       Request finished in 985.8655ms 200 application/json

wilriker, 2019-07-22:
Which version of `systemd` is running on Raspbian?

beikeland, 2019-07-22:
Now thats a fine question, so I asked google;)
```
pi@duet3:~ $ dpkg -l systemd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name      Version         Architecture Description
+++-=========-===============-============-==========================
ii  systemd   232-25+deb9u11  armhf        system and service manager
```

chrishamm, 2019-07-22:
Can you either reboot your Pi or run "sudo systemctl daemon-reload" and check if that makes a difference? I'll try to improve the packages further this week.

wilriker, 2019-07-22:
OK, Arch Linux is running `242.32` which is significantly newer. Not sure though if that makes a difference. I will have to try further.

@chrishamm The Pi has been rebooted so many times and also `pacman` has built-in hooks where it detects if a package contains a `systemd` unit file and reloads the daemon in case it has been changed.

wilriker, 2019-07-22:
But I think there is something missing in the unit file either way (though it does not help adding it). It should list `duetcontrolserver` not just as `Wants` but also as `Requires` since it does not really work if `DCS` is not running:

```systemd
[Unit]
Description=Duet Web Server
Wants=duetcontrolserver.service
Requires=duetcontrolserver.service
After=network.target duetcontrolserver.service

[Service]
ExecStart=/opt/dsf/bin/DuetWebServer
TimeoutStopSec=15
Type=simple

[Install]
WantedBy=multi-user.target
```

wilriker, 2019-07-23:
I am going to close this now because on a freshly installed Arch Linux on my new RPi 4B it can be started and run as systemd unit. No idea what was blocking it on my other RPi but it seems to be something in that specific installation (or it was fixed in the systemd update to 242.84 yesterday). Either way it works.

wilriker, 2019-07-23:
I have to reopen this issue because I now seem to have found the root cause:
Everything works as long as I do not have a NFS-share mounted. If I do that then DuetWebServer seems to want to index every file in it. If I run `lsof | grep <mountpoint>` I see an immense amount of handles that are held by DuetWebServer. On a successful start (i.e. without NFS share mounted) I can see the following line in the log output:
```
Jul 23 10:53:14 duet3 DuetWebServer[715]: Content root path: /
```

Extending the `duetwebserver.service` file to look like
```
[Unit]
Description=Duet Web Server
Wants=duetcontrolserver.service
After=network.target duetcontrolserver.service

[Service]
WorkingDirectory=/opt/dsf/sd
ExecStart=/opt/dsf/bin/DuetWebServer
TimeoutStopSec=15
Type=simple

[Install]
WantedBy=multi-user.target
```
1) fixes the issue and
2) speeds up start of DWS by a factor of 2.
3) let's DWS use a lot less RAM since it does not have to cache all files available to the complete system

chrishamm, 2019-07-26:
Interesting, I would not have thought that the working directory makes a change since files are only resolved from the www directory. Thanks, I will change it in the next version.

wilriker, 2019-07-26:
Seems to me that kestrel wants to be "prepared". :-D

wilriker, 2019-07-26:
> since files are only resolved from the www directory. 

In that case setting the working directory to this dir would even be better instead of using the whole virtual SD card. The fewer it has to index the faster and more lightweight it is.

wilriker, 2019-08-05:
Fixed in latest package for `DWS`.

closed 2019-08-05 by wilriker