Duet3D/DuetSoftwareFramework #30: DWC not honoring jog button speed

wilriker, 2019-07-21:
`DCS`: 1.0.2.1
`DWS`: 1.0.1.1
`DWC`: 2.0.0-RC8-b1

Not sure if this belongs here or to DWC.
The jog buttons will run GCode with `F18000` (a value I set at some time in the past on Duet 2 and I copied the DWC settings file over). Channing the value in DWC Machine-specific settings tab does not affect which value is used. The changed value is persisted to dwc2settings.json, though.

wilriker, 2019-07-23:
I think this rather belongs to DWC. I just found out that it treats the 
```
Feedrate for move buttons (mm/min)
``` actually as `mm/s`. If I lower this down to 5 then I get the aforementioned 300mm/min.

chrishamm, 2019-07-24:
Thanks, will fix that soon

chrishamm, 2019-08-07:
Fixed in the latest DWC beta.

closed 2019-08-07 by chrishamm