Duet3D/DuetSoftwareFramework #31: M115 does not work on serial/USB

beikeland, 2019-07-21:
Wanted to grab the FW version from `M115` off the serial, but when sending `M115` the response is `Emulation of unknown is not yet supported.` `M115`  works as expected in Web UI.

beikeland, 2019-07-22:
FYI, I just wanted to make it easier and more consistent to report what version is being tested against, and to rule out human error in the process. Came up with a workaround based off the currently loaded firmware binary

if of interest:
```
pi@duet3:~ $ ./versions.sh
duetcontrolserver=1.0.2.1
duetruntime=1.0.2.1
duetsd=1.0.1
duetsoftwareframework=1.0.2.1
duettools=1.0.2.1
duetwebcontrol=2.0.0-2
duetwebserver=1.0.1.1
duet3firmware=3.0b6-ch@2019-07-21b1
pi@duet3:~ $ cat versions.sh
#!/bin/sh
FW=/opt/dsf/Duet3Firmware.bin
FWVER=`strings $FW | grep version= | awk -F= '{print $2}'`
FWDATE=`strings $FW | grep FIRMWARE_DATE -b1 |head -n1 | awk -F- 'BEGIN {OFS="-"} { $1=""; print $0}' | cut -b2-`
SWVER=`apt list 2>/dev/null | grep duet | awk -F'[/ ]' '{print $1"="$3}'`
echo $SWVER | tr ' ' '\n'
echo "duet3firmware=$FWVER@$FWDATE"
```

wilriker, 2019-07-22:
@beikeland Nice idea. But I think when fetching the `FIRMWARE_DATE` you wanted to use `grep -B1 FIRMWARE_DATE` (mind the capital `B`). This will also relief you from using `awk` and everything that follows. :-)

beikeland, 2019-07-22:
True story!

Thanks, edited the post above to reflect your input.

wilriker, 2019-07-22:
Also here is a version that runs on Arch Linux
```bash
#!/usr/bin/env bash

pacman -Qs duet | grep "local/" | sed 's#local/##g' | tr ' ' =

FW=/opt/dsf/sd/sys/Duet3Firmware.bin
FWVERSION=$(strings "${FW}" | grep "version=" | cut -d= -f2)
FWDATE=$(strings "${FW}" | grep -B1 "FIRMWARE_DATE" | head -1)
echo "duet3firmware=${FWVERSION}@${FWDATE}"
```
I also removed `awk` in my version completely (I just don't like it) and I am pointing to the firmware file that was uploaded through `DWC` most recently.

wilriker, 2019-07-22:
If you are running a regular RPi you should look into setting up `rngd` to use the hardware RNG that is on-board.
```
/usr/bin/rngd -f -o /dev/random -r /dev/hwrng
```

beikeland, 2019-07-22:
The good thing about wise-guys is that it makes us non-wise guys less non-wise ;)

Seriously, thanks for sharing!

Did away with an awk as well, when all you have in your toolbox is a hammer, the scews feel the pain..

beikeland, 2019-07-22:
Since this is already waay off topic; simply installing rng-tools seems to have sorted the entropy deficiency on the pi, and its swiftly updated on boot flashing and notifying me as needed. Yay!

Beats me why the rng-tools package isn't installed by default, but thats a topic for another place, thanks again!

beikeland, 2019-07-22:
Still slightly unrelated to the actual issue, but think the serial and http part is sorted, should increase the odds of reporting the actual running firmware version

wilriker, 2019-07-24:
Took me a while to get at actually testing the initial issue and I cannot reproduce it:
```
$ miniterm.py --echo /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
M115
FIRMWARE_NAME: RepRapFirmware for Duet 3 v0.5 FIRMWARE_VERSION: 3.0b6-ch ELECTRONICS: Duet 3 prototype v0.5 FIRMWARE_DATE: 2019-07-21b1
```

chrishamm, 2019-08-07:
I think this has been fixed in a recent RRFv3 version. There was an error concerning emulation.

closed 2019-08-07 by chrishamm

beikeland, 2019-08-08:
Confirmed; works as of duet3_firmware=3.0b7-ch@2019-08-07b2