Duet3D/DuetSoftwareFramework #32: Packages file only has latest version

beikeland, 2019-07-22:
Probably not a priority but https://chrishamm.io/debian/dists/stretch/dsf/binary-armhf/Packages seems to only contain the latest version, and not refrences to all the available .deb packages preventing apt to downgrade.

beikeland, 2019-08-05:
Stretch packages are no longer built? I notice the buster Packages file has newer versions but the Stretch does not, intentional?

chrishamm, 2019-08-07:
I just updated the stretch packages, although I think the current DSF version (1.0.3.1) will be the last one I will host on my private server. Once the location has changed, I will update the Wiki accordingly.

chrishamm, 2019-09-03:
Please see https://duet3d.dozuki.com/Wiki/Duet_3_prototype_guide_for_OEMs#Section_Installation - I just updated the instructions for the new official package feeds. There will be no more feeds for individual distro versions like stretch or buster.

closed 2019-09-03 by chrishamm