Duet3D/DuetSoftwareFramework #33: Failed to add /run/systemd/ask-password to directory watch: No space left on device

beikeland, 2019-07-22:
```
pi@duet3:~ $ sudo /bin/systemctl stop duetwebserver
Failed to add /run/systemd/ask-password to directory watch: No space left on device
```

Webserver seems to be stopped, but this message is output (occasionally?). Have been an issue with all versions tested so far. 

Plain Debian Stretch (lite) with screen and rng-tools as only non standard software.

wilriker, 2019-07-22:
Not sure how this is done in Debian but on Arch Linux `/run` is of type `tmpfs` (i.e. similar to a ramdisk) and as such shares space with all other `tmpfs`-typed mounts. The only one really using up space usually is `/tmp`. Can you check with
```
df -h
```
how much space is available and if my assumption is correct that `/tmp` is holding all the space can you also check what uses all the space in there?

beikeland, 2019-07-22:
similar, and there is plenty of space
```
pi@duet3:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  1.9G   13G  14% /
devtmpfs        459M     0  459M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   12M  452M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   44M   23M   21M  52% /boot
tmpfs            93M     0   93M   0% /run/user/1000
``` 
Google seems to suggest it could be related to number of inodes being able to be watched, but not my area of expertise. I just find it strange it only applies to the webserver. Its not an actual problem, but maybe something worth being aware of before it turns into a problem when released to the masses.

wilriker, 2019-07-22:
Can you check with
```
cat /proc/sys/fs/inotify/max_user_watches
```
how many watches are configured and with
```
sudo lsof -K | grep inotify | less
```
(you might need to install `lsof`) what uses up all of these? I am no where near the limit (on my machine 512K watches allowed vs. ~30 watches active).

beikeland, 2019-07-22:
Less configured, but about same amount used (varied around 30-40-ish), mostly DuetWebServer. And should be plenty free.
```
pi@duet3:~ $ cat /proc/sys/fs/inotify/max_user_watches
8192
```

```
$ sudo lsof -K | grep inotify
sd-resolv  280  326 systemd-timesync   11r  a_inode       0,11        0       5804 inotify
gmain     2416 2417             root    8r  a_inode       0,11        0       5804 inotify
gmain     2416 2417             root    9r  a_inode       0,11        0       5804 inotify
gdbus     2416 2419             root    8r  a_inode       0,11        0       5804 inotify
gdbus     2416 2419             root    9r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5337             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5337             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5338             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5338             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5339             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5339             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5340             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5340             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5341             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5341             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5342             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5342             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5343             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5343             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5344             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5344             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5345             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5345             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5350             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5350             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5353             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5353             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5354             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5354             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5359             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 5359             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 9660             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 9660             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 9881             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 9881             root   99r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 9917             root   95r  a_inode       0,11        0       5804 inotify
DuetWebSe 5333 9917             root   99r  a_inode       0,11        0       5804 inotify
```

wilriker, 2019-07-22:
Wow, 8K that is... crazy low default value. Still, as you said, this should be plenty.

I have not yet seen this message on my Arch Pi but then again I cannot run DuetWebServer as systemd unit in the first place. Strange. Now I stop spamming. ;-)

beikeland, 2019-07-22:
Oddly enough raising the limit to 512k seems to have a positive effect. Will persist the change and see if it the effect persists.

`sudo bash -c 'echo "fs.inotify.max_user_watches=524288" >> /etc/sysctl.conf'`

It's not spam when it's educational ;) :)

wilriker, 2019-07-23:
This might be related to my latest findings in #29.

chrishamm, 2019-08-07:
Should be fixed in DWS 1.1.0.0 - it's probably related to the amount of inotify items.

closed 2019-08-07 by chrishamm