Duet3D/DuetSoftwareFramework #35: Connection issues behind reverse proxy

wilriker, 2019-07-22:
duetcontrolserver=1.0.2.1-1
duetruntime=1.0.2.1-1
duetsd=1.0.1-2
duetsoftwareframework-meta=1.0.2.1-1
duettools=1.0.2.1-1
duetwebcontrol=2.0.0.RC8.b1-1
duetwebserver=1.0.1.1-1
duet3firmware=3.0b6-ch@2019-07-21b1

This is new with the latest release. I get lots of connection
```
Connection interrupted, attempting to reconnect...
Network error
```
when connecting through a reverse proxy (`nginx`) to load `DWC`. This was working without any issues before.

chrishamm, 2019-07-24:
Perhaps this helps re DWS: https://docs.microsoft.com/en-us/aspnet/core/host-and-deploy/linux-nginx?view=aspnetcore-2.2 I have yet not tried to set it up as a reverse proxy, though.

Do you get any messages in the DWS log?

wilriker, 2019-07-24:
I have my reverse proxy setup exactly as described in the article (I even add some more additional `X-Forwarded-` headers).

Again, it worked correctly with DWS 1.0.0.3 and started to be an issue with DWS 1.0.1.1. But maybe this is also somehow related to #27.

I will check for DWS log messages tomorrow.

wilriker, 2019-07-25:
So here is what I see in the DWS log
```
Jul 25 10:15:03 duet3 DuetWebServer[265]: fail: DuetWebServer.Controllers.MachineController[0]
Jul 25 10:15:03 duet3 DuetWebServer[265]:       WebSocket terminated with an exception
Jul 25 10:15:03 duet3 DuetWebServer[265]: System.Net.WebSockets.WebSocketException (2): The remote party closed the WebSocket connection without completing the close handshake. ---> System.Net.WebSockets.WebSocketException (2): The remote party closed the WebSocket connec
tion without completing the close handshake.
Jul 25 10:15:03 duet3 DuetWebServer[265]:    at System.Net.WebSockets.ManagedWebSocket.ThrowIfEOFUnexpected(Boolean throwOnPrematureClosure)
Jul 25 10:15:03 duet3 DuetWebServer[265]:    at System.Net.WebSockets.ManagedWebSocket.EnsureBufferContainsAsync(Int32 minimumRequiredBytes, CancellationToken cancellationToken, Boolean throwOnPrematureClosure)
Jul 25 10:15:03 duet3 DuetWebServer[265]:    at System.Net.WebSockets.ManagedWebSocket.ReceiveAsyncPrivate[TWebSocketReceiveResultGetter,TWebSocketReceiveResult](Memory`1 payloadBuffer, CancellationToken cancellationToken, TWebSocketReceiveResultGetter resultGetter)
Jul 25 10:15:03 duet3 DuetWebServer[265]:    at DuetWebServer.Controllers.WebSocketController.Process(WebSocket webSocket, String socketPath, ILogger logger) in /home/christian/duet/DuetSoftwareFramework/src/DuetWebServer/Controllers/WebSocketController.cs:line 67
```

wilriker, 2019-07-25:
Some additional information: connection usually breaks up after about 60 seconds occasionally lasting up to 110 seconds. And this seems independent of whether I actually have DWC tab open in the browser or if it's currently a background tab. Also it happens e.g. if I take "too long" to edit `config.g`.

wilriker, 2019-07-25:
And I think I found the source in this blog post: https://blog.carbocation.com/post/51529889500/how-to-fix-nginx-websockets-that-disconnect-after

And this makes it definitely a side-effect of #27 since there are no more continuous updates `nginx` will close the websocket. I will test if the above solution helps and if so close this issue.

wilriker, 2019-07-25:
OK, so the workaround given in the linked blog post does solve the permanent disconnects. But I think this also reveals that the new PING-PONG does not really work yet.

Following is what I can see was sent on the websocket over the last 15 minutes (15:51-16:06) with PING-PONG interval being set to the default of 2000ms.
```
<DATA> | 10988 | 15:51:20.996
OK | 3 | 15:51:21.578
<DATA>| 211 | 15:55:29.099
OK | 3 | 15:55:29.147
PING | 5 | 15:55:31.148
<DATA> | 197 | 16:00:18.553
OK | 3 | 16:00:18.580
PONG | 5 | 16:00:18.583
PING | 5 | 16:00:20.580
```

Should I close this one and open a new issue for PING-PONG not working correctly? And if so should I open this issue here or over at DWC?

chrishamm, 2019-07-26:
Thanks for the observation! I just implemented a fix and will roll it out as part of the next DSF update.

wilriker, 2019-07-29:
This is fixed in `DCS` 1.0.2.2.

wilriker, 2019-08-05:
Reopening this because though this is **implicitly** fixed in 1.0.2.2 it only really is a workaround because the frequency of transmitting data was increased. In a case where in the future out of any reason the amount of data will be reduced a lot again this will re-occur.

Kestrel already sends WebSocket Ping frames but it only does so in 120 second intervals whereas `nginx` will consider the connection to be dead after 60 seconds already. My recommendation would be to lower the interval in Kestrel to 30 seconds as described at
https://docs.microsoft.com/en-us/aspnet/core/fundamentals/websockets?view=aspnetcore-2.2#configure-the-middleware

Also it might be beneficial to configure the use of forwarded headers as described at https://docs.microsoft.com/en-us/aspnet/core/host-and-deploy/linux-nginx?view=aspnetcore-2.2#use-a-reverse-proxy-server

wilriker, 2019-08-05:
And this has just returned with `DCS` 1.0.3.0 together with #27.

wilriker, 2019-08-05:
I can confirm Kestrel ping frames to work. I meanwhile set my `nginx` to timeout after 300 seconds and it keeps the connection, i.e. Kestrel sending ping frames every 120 seconds seems to work.

Though as mentioned in #27 there is no data sent through the WebSocket after the initial message anymore.

chrishamm, 2019-08-07:
I reckon the PING/PONG mechanism used by DWC2 to detect connection interrupts from the client-side should be able to keep the connection alive, too (interval = 2s). Since you mentioned there was a bug in this mechanism. I fixed it and if you upgrade to DWC2-RC8b2 or later, this problem should be gone.

wilriker, 2019-08-09:
There still seems to be a bug in DWC2 PING/PONG. I explicitly updated DWC first leaving everything else as it was, i.e. no status updates with a configured tool. In this case there still was no PING/PONG on the `machine` WebSocket. There was only the initial message and the corresponding "OK". Nothing else.

Updating all other components got the updates flowing again so I cannot reproduce this anymore. But it would be possible by downgrading `DCS` and probably `DWS` as well.

Also IMHO PING/PONG in DWC2 is a bit of reinventing the wheel since Kestrel already implements the WebSocket ping frames and I can confirm that they do work.

chrishamm, 2019-08-09:
That's perfectly fine as long as the socket doesn't idle for more than 2 seconds (or whatever you ping interval is set to). The whole mechanism is only used to detect connection aborts so if DWC2 receives messages in intervals less than the ping interval, no PING/PONG messages are exchanged.

The main reason for the new PING/PONG approach is the fact that DWC2 would remain connected when the Pi powers down (at least with Firefox, I haven't tested it with Chrome yet). AFAIR there is no way to set connection keep-alive/timeout for WebSockets.

wilriker, 2019-08-09:
> That's perfectly fine as long as the socket doesn't idle for more than 2 seconds (or whatever you ping interval is set to). The whole mechanism is only used to detect connection aborts so if DWC2 receives messages in intervals less than the ping interval, no PING/PONG messages are exchanged.

But that was not the case as I described. There were no messages after the initial status message and the corresponding OK on the WebSocket for far more than 2 seconds. DWC2 did not send any PING and thus never received a PONG. Of course this can only be seen if the continuous status updates coming from `DCS` are broken (as in 1.0.3.0 with a tool configured).

chrishamm, 2019-08-09:
I just checked and it does send PING after 2 seconds of inactivity (DWC 2.0.0-RC8b3).

wilriker, 2019-08-09:
How do you "produce" this inactivity?

chrishamm, 2019-08-09:
Cut the power to the RaspPi.

I reckon it's still broken in case only a single initial message is received so I'll add a fix to the next DWC2 version. Thanks for reporting that.

chrishamm, 2019-09-03:
Fixed in DSF 1.0.3.3 (unstable).

closed 2019-09-03 by chrishamm