Duet3D/DuetSoftwareFramework #36: DCS not connecting to RRF on RPi 4B

wilriker, 2019-07-22:
duetcontrolserver=1.0.2.1-1
duetruntime=1.0.2.1-1
duetsd=1.0.1-2
duetsoftwareframework-meta=1.0.2.1-1
duettools=1.0.2.1-1
duetwebcontrol=2.0.0.RC8.b1-1
duetwebserver=1.0.1.1-1
duet3firmware=3.0b6-ch@2019-07-21b1

I received my RPi 4B today and right now I cannot use it to connect DCS to RRF. It just states "Duet is not available" and exits.

Is there any config flag to make it more verbose about this or do I need to add more logging to DCS to find out what is going wrong?

I know that instructions ask for a RPi 3B+ but since it was also working with my RPi 3B (without the plus) and 4B has the same GPIO layout I assumed it would also work out of the box.

T3P3, 2019-07-22:
Possibly due to the difference in SPI setup?

wilriker, 2019-07-22:
Maybe. Though with the very limited information from the error message it is hard to say.

I do get the same (named) devices for SPI under `/dev` at least. And they have the same kernel-signed major, minor device IDs.

wilriker, 2019-07-22:
Found the very likely source: `wiringPi` does not yet support the RPi 4B.

chrishamm, 2019-07-22:
I already tried to use WiringPi 2.52 on my RaspPi 4B but unfortunately the API is still buggy: Even though their gpio program reports correct values, it looks like their API library provides wrong GPIO values so DCS always thinks the transfer ready pin is low although it is actually high.

That said, don't bother to try it out tomorrow. If it helps, I can update this issue once I get my RaspPi 4 working with DSF.

beikeland, 2019-07-23:
Is the 4B running Stretch or Buster?

wilriker, 2019-07-23:
I already tried last night with 2.52 and can confirm that `gpio` reports pin `GPIO.6` to be high but still `DCS` is not connecting.

@beikeland Mine of course runs Arch Linux. ;-)

wilriker, 2019-07-23:
Further tracked down the issue. It's `Unosquare.WiringPi.Native`. AFAICT it has hard-coded references to `libwiringPi.so.2.46`.

wilriker, 2019-07-23:
And until this is fixed upstream here is a working version of `Unosquare.WiringPi.dll` that I just compiled with the latest `libwiringPi.so.2.52`:
https://www.dropbox.com/sh/vaqrk83s3d19173/AACUf_7nwygZirFfMvRZ3YW1a?dl=0

This one works on RPi 4B and most probably also on all older versions.

I close this now.

closed 2019-07-23 by wilriker

wilriker, 2019-07-23:
P.S.: This needs `wiringPi` 2.52 installed in the system.