Duet3D/DuetSoftwareFramework #37: DWC requires reload after board reset

T3P3, 2019-07-23:
duetcontrolserver 1.0.2.2
duetruntime 1.0.2.1
duetsoftwareframework 1.0.2.1
duetwebcontrol 2.0.0-2
duetwebserver 1.0.1.1

Issue: The Extrude/Retract buttons do not work in DWC after an emergency stop until DWC is reloaded (normally by refreshing the page using F5)

How to recreate: Use emergency stop to reset the Duet. Once the reset has completed ensure the hotend is up to temperature, then prese Extruder/Retract. No commands will be sent to DCS (and DWS does not report the button presses). Refresh the page (F5) try Extrude/Retract again. The commands are now sent to DWS->DCS.

wilriker, 2019-07-23:
I can confirm this.

chrishamm, 2019-08-07:
Fixed in DCS 1.0.3.1.

closed 2019-08-07 by chrishamm