Duet3D/DuetSoftwareFramework #38: Date and time on Duet not initialized

wilriker, 2019-07-23:
Previously DWC would send the current date and time when connecting via `rr_connect?password=xxx&time=...` but this does not happen anymore. So now the Duet won't get any more information about current date and time.

This could be implemented to happen upon successful connection of DCS to RRF via [`M905`](https://duet3d.dozuki.com/Wiki/Gcode?revisionid=HEAD#Section_M905_Set_local_date_and_time).

chrishamm, 2019-07-24:
AFAIR the datetime is only used for FS timestamps in RRF and delivering the current time makes sense on a Duet 2 because unlike an SBC these boards don't posess an NTP client. So unless it is used for something else in RRF, I believe it makes sense to drop support for M905.

wilriker, 2019-07-24:
The least important part where it is used is in the output of `M122`. But I think it is also used in logging. @dc42 could tell more about that.

dc42, 2019-07-26:
The date/time is needed by RRF both for logging (although as logging will go via the RPi, it could be added to the log messages by DSF) and also to record the time of the last software reset, which is useful to know when diagnosing problems.

wilriker, 2019-08-05:
Implemented in `DCS` 1.0.3.0.

closed 2019-08-05 by wilriker