Duet3D/DuetSoftwareFramework #39: DWC/DCS now extremely slow to respond

T3P3, 2019-07-23:
Since the versions that included buffering i have noticed a very significant slowdown in DWC response, which shows as a very slow rate of gcodes being sent by DCS. Response times are now so laggy that its usual for me to have multiple different commands sent by DWC "queued up" (i.e. i have pressed a button i am used to completing instantly, like set temperature) and because it has taken so long to complete i have sent more commands before it completes. This has caused crashes:
```
info] Completed G10 P0 S200
[warn] Handled exception: System.InvalidOperationException: An attempt was made to transition a task to a final state when it had already completed.
   at System.Threading.Tasks.TaskCompletionSource`1.SetResult(TResult result)
   at DuetControlServer.SPI.QueuedCode.SetFinished() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/QueuedCode.cs:line 132
   at DuetControlServer.SPI.QueuedCode.HandleReply(MessageTypeFlags messageType, String reply) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/QueuedCode.cs:line 99
   at DuetControlServer.SPI.ChannelInformation.Invalidate(String codeErrorMessage) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Channel.cs:line 533
   at DuetControlServer.SPI.Interface.InvalidateData(String codeErrorMessage) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Interface.cs:line 755
   at DuetControlServer.SPI.Interface.RequestEmergencyStop() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Interface.cs:line 173
   at DuetControlServer.Codes.MCodes.Process(Code code) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Codes/MCodes.cs:line 227
   at DuetControlServer.Commands.Code.ExecuteInternally() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Commands/Code.cs:line 147
[info] Completed M140 P0 S60
[info] Completed T0
[info] Completed M122
System.AggregateException: One or more errors occurred. (An attempt was made to transition a task to a final state when it had already completed.) ---> System.InvalidOperationException: An attempt was made to transition a task to a final state when it had already completed.
   at System.Threading.Tasks.TaskCompletionSource`1.SetResult(TResult result)
   at DuetControlServer.SPI.QueuedCode.SetFinished() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/QueuedCode.cs:line 132
   at DuetControlServer.SPI.QueuedCode.HandleReply(MessageTypeFlags messageType, String reply) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/QueuedCode.cs:line 99
   at DuetControlServer.SPI.ChannelInformation.Invalidate(String codeErrorMessage) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Channel.cs:line 533
   at DuetControlServer.SPI.Interface.InvalidateData(String codeErrorMessage) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Interface.cs:line 755
   at DuetControlServer.SPI.Interface.RequestEmergencyStop() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Interface.cs:line 173
   at DuetControlServer.Codes.MCodes.Process(Code code) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Codes/MCodes.cs:line 227
   at DuetControlServer.Commands.Code.ExecuteInternally() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Commands/Code.cs:line 147
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at DuetControlServer.Commands.Code.Enqueue() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Commands/Code.cs:line 109
   at DuetControlServer.Commands.SimpleCode.Execute() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Commands/SimpleCode.cs:line 31
   at DuetAPI.Commands.Command`1.Invoke() in /home/christian/duet/DuetSoftwareFramework/src/DuetAPI/Commands/BaseCommand.cs:line 86
   at DuetControlServer.IPC.Processors.Command.Process() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/IPC/Processors/Command.cs:line 61
---> (Inner Exception #0) System.InvalidOperationException: An attempt was made to transition a task to a final state when it had already completed.
   at System.Threading.Tasks.TaskCompletionSource`1.SetResult(TResult result)
   at DuetControlServer.SPI.QueuedCode.SetFinished() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/QueuedCode.cs:line 132
   at DuetControlServer.SPI.QueuedCode.HandleReply(MessageTypeFlags messageType, String reply) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/QueuedCode.cs:line 99
   at DuetControlServer.SPI.ChannelInformation.Invalidate(String codeErrorMessage) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Channel.cs:line 533
   at DuetControlServer.SPI.Interface.InvalidateData(String codeErrorMessage) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Interface.cs:line 755
   at DuetControlServer.SPI.Interface.RequestEmergencyStop() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/SPI/Interface.cs:line 173
   at DuetControlServer.Codes.MCodes.Process(Code code) in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Codes/MCodes.cs:line 227
   at DuetControlServer.Commands.Code.ExecuteInternally() in /home/christian/duet/DuetSoftwareFramework/src/DuetControlServer/Commands/Code.cs:line 147<---
```
and out of sync errors

`"[warn] Received out-of-sync code reply (HttpMessage, BinaryCodeReplyFlag: )"`

it is also makes the UI highly frustrating to use.

I only noticed this on the most recent versions of DCS (1.0.1.0 and later) so it don't think its an issue with the underlying SPI speed (i have this set to 5Mhz). is DCS/RRF waiting for a buffer to be full before sending it?

T3P3, 2019-07-23:
Possibly related i am seeing these occasionally:
[warn] Restarting transfer because a bad header response was received (0x0001015f)

wilriker, 2019-07-23:
I think
```
[warn] Restarting transfer because a bad header response was received (0x0001015f)
```
has more to do with #22. I see that if I limit the memory to a very low limit and it tries to continue.

Other than that I just tried clicking on a jog button to move the X axis as fast as I possibly can and it worked flawlessly.

T3P3, 2019-07-23:
how about the temperature change drop down, or the homeall button?

Maybe its an Raspian issue? Looking at the memory usage:
KiB Mem :   948304 total,    68388 free,   220940 used,   658976 buff/cache its not so desperate

wilriker, 2019-07-23:
No lag on my side. Sometimes it just fails to finish commands, though, and then the UI just has a spinner somewhere that does not disappear.

Where do you access DWC from? A PC or on a screen connected to the RPi using `localhost`?

wilriker, 2019-07-23:
Not printing anything but I get
```
=== Duet Control Server ===
Duet Control Server v1.0.2.1
Code buffer space: 1024
SPI speed: 1000000 Hz
Full transfers per second: 25.53
```

T3P3, 2019-07-23:
over the network normally but its just as laggy on the Pi directly - possibly even more laggy as chromium is now stressing it out!

wilriker, 2019-07-23:
Are you maybe confusing the lag with #27? It does not update the UI in most cases for me at all. I usually have to reload the page to see any changes. I e.g. just typed `G92 X-10` and it showed X to be homed only after page reload.

wilriker, 2019-07-23:
I had to take the time: I just was able to click the jog button 75 times in 8 seconds with no delay in command execution or (in this case available) UI feedback.

chrishamm, 2019-07-26:
I could track this down to the used SPI/GPIO driver. I am really fed up with the quality of external GPIO libraries for the Pi so I will implement my own driver classes today. Once they are complete, people will be able to use DSF on non-Raspberry boards, too.

chrishamm, 2019-07-29:
Will be fixed in the next version.

chrishamm, 2019-08-07:
Fixed in DCS 1.0.3.1.

closed 2019-08-07 by chrishamm