Duet3D/DuetSoftwareFramework #41: DCS does max out one CPU core

wilriker, 2019-07-29:
duetcontrolserver=1.0.2.2

As the title says, `DCS` does max out one of the CPU cores (96%) of my RPi 4B just with status updates. If no client is connected, i.e. `DWS` stopped it still uses 50-60% of one CPU core. This is a lot of resources for basically "doing nothing".

chrishamm, 2019-07-29:
That is related to the object model (RRF status responses). It will be improved quite soon which will notably reduce the CPU usage.

wilriker, 2019-07-29:
I thought so but wanted to document it here either way.

wilriker, 2019-08-05:
This is reduced considerably in `DCS` 1.0.3.0. Closing.

wilriker, 2019-08-07:
Reopening. This has not been improved. It only seemed to be better because of #27 that mainly is caused by #47.

If no tool is specified and therefore status updates are working idle load (i.e. with `DWS` stopped and no clients connected to `DCS`) is still between 50-60% and load with a client connected is slightly lowered to an average of around 85%.

chrishamm, 2019-08-07:
The latest DCS version performs a relatively high number of SPI transactions per second when G-codes are being executed, which explains the CPU load. Although DCS/RRF now support code buffering, I am not willing to introduce more delays yet. I will do this at a later point when I find more time for fine-tuning, which will eventually reduce the CPU usage. Besides, I expect another way of object model transfers (binary vs JSON) and the planned upgrade to .NET Core 3 will further lower the CPU load. It may take a while before these points are addressed, though.

wilriker, 2019-08-09:
In `DCS` 1.0.3.1 I finally see a considerable reduction in CPU usage while updates are flowing (<20%) and even lower (6-10%) if no client is connected. This is a very much acceptable resource usage. I am going to close this now.

In case this was not obvious: I was all the time talking about the printer being idle, i.e. not performing any task. I don't mind if the CPU usage increases if the printer is actually doing something.

Seeing that this will even reduce further in the future I will close this issue. Thanks!

closed 2019-08-09 by wilriker