Duet3D/DuetSoftwareFramework #43: Inconsistent file ownership in virtual SD folder

beikeland, 2019-07-29:
After trying to edit some of the config files from the ssh shell I notice ownership is inconsistent.

config.g is owned by root, the other by pi. I change all ownership to pi.
```
pi@duet3:~ $ ls -lh  /opt/dsf/sd/sys/
total 64K
-rw-r--r-- 1 root root  614 Jul 29 12:14 config.g
-rw-r--r-- 1 root root  613 Jul 29 11:36 config.g.bak
-rw-r--r-- 1 pi   pi   3.1K Jul 22 16:05 dwc2settings.json
-rw-r--r-- 1 pi   pi    521 Jul 29 12:07 homeall.g
-rw-r--r-- 1 pi   pi    486 Jul 29 12:17 homex.g
-rw-r--r-- 1 pi   pi    625 Jul 28 11:48 homey.g
-rw-r--r-- 1 pi   pi    383 Jul 28 11:46 homez.g
-rwxr-xr-x 1 pi   pi    34K Jul 21 09:15 iapduet3.bin
pi@duet3:~ $ sudo chown -R pi:pi  /opt/dsf/sd/sys/
pi@duet3:~ $ ls -lh  /opt/dsf/sd/sys/
total 64K
-rw-r--r-- 1 pi pi  614 Jul 29 12:14 config.g
-rw-r--r-- 1 pi pi  613 Jul 29 11:36 config.g.bak
-rw-r--r-- 1 pi pi 3.1K Jul 22 16:05 dwc2settings.json
-rw-r--r-- 1 pi pi  521 Jul 29 12:07 homeall.g
-rw-r--r-- 1 pi pi  486 Jul 29 12:17 homex.g
-rw-r--r-- 1 pi pi  625 Jul 28 11:48 homey.g
-rw-r--r-- 1 pi pi  383 Jul 28 11:46 homez.g
-rwxr-xr-x 1 pi pi  34K Jul 21 09:15 iapduet3.bin
```
Now I edit config.g and homeall.g in DWS and check again:
```
pi@duet3:~ $ ls -lh  /opt/dsf/sd/sys/
total 64K
-rw-r--r-- 1 root root  613 Jul 29 12:34 config.g
-rw-r--r-- 1 pi   pi    614 Jul 29 12:14 config.g.bak
-rw-r--r-- 1 pi   pi   3.1K Jul 22 16:05 dwc2settings.json
-rw-r--r-- 1 pi   pi    521 Jul 29 12:35 homeall.g
-rw-r--r-- 1 pi   pi    486 Jul 29 12:17 homex.g
-rw-r--r-- 1 pi   pi    625 Jul 28 11:48 homey.g
-rw-r--r-- 1 pi   pi    383 Jul 28 11:46 homez.g
-rwxr-xr-x 1 pi   pi    34K Jul 21 09:15 iapduet3.bin
```
Only config.g has changed owner to root again, not homeall.g

wilriker, 2019-07-29:
The reason is most likely that `config.g` gets first moved to `config.g.bak` when being saved. Next a new file `config.g` is created. And since `DWS` runs as `root` it will create the new file as root. `homeall.g` will be overwritten in place and as such keeps its ownership.

chrishamm, 2019-08-07:
Thanks, I will have a look at this soon. Since DCS/DWS both run as 'root', I don't think this will cause any serious problems, though.

chrishamm, 2019-10-08:
There is barely a reason this would cause conflicts so I am closing it. Perhaps it would be possible to run DCS/DWS as user 'pi' but I have not tested that yet. In any case, it would require access at least to the GPIO and SPI device nodes and I am not 100% sure if user 'pi' has that without extra configuration.

closed 2019-10-08 by chrishamm