Duet3D/DuetSoftwareFramework #46: DCS blocks while executing start.g

wilriker, 2019-08-02:
duetcontrolserver=1.0.2.2-1
RRF=3.0b7-ch@2019-07-26b1

When trying to print something `DCS` will send the very first line of `start.g` and then block completely. It will no longer send status updates  to `DWS` and will not react to Emergency Stop button clicks in `DWC`. `DCS` has to be restarted in order to interact with it again.

Log excerpt
```

Aug 02 13:12:13 duet3 DuetControlServer[2723]: Duet Control Server v1.0.2.2
Aug 02 13:12:13 duet3 DuetControlServer[2723]: Written by Christian Hammacher for Duet3D
Aug 02 13:12:13 duet3 DuetControlServer[2723]: Licensed under the terms of the GNU Public License Version 3
Aug 02 13:12:15 duet3 DuetControlServer[2723]: Loading settings... Done!
Aug 02 13:12:15 duet3 DuetControlServer[2723]: Initialising object model... Done!
Aug 02 13:12:15 duet3 DuetControlServer[2723]: Connecting to RepRapFirmware... Done!
Aug 02 13:12:15 duet3 DuetControlServer[2723]: Creating IPC socket... Done!
Aug 02 13:12:47 duet3 DuetControlServer[2723]: [info] Printing file '/opt/dsf/sd/gcodes/topsurface.gcode'
Aug 02 13:12:47 duet3 DuetControlServer[2723]: [info] Executing system macro file '/opt/dsf/sd/sys/start.g' on channel File
Aug 02 13:12:47 duet3 DuetControlServer[2723]: [info] Sent M80 ; Turn the VIN PSU on, remaining space 748 (2028 total), needed 20
```
Nothing happens then anymore until restart of `DCS`.

Also after having restarted `DCS` if the heaters were turned on before they will all be `off` then. This is **not** the case if `DCS` is restarted while it is still working.

chrishamm, 2019-08-02:
Thanks, I could reproduce this problem and will fix it.

chrishamm, 2019-08-07:
Fixed in DCS 1.0.3.1.

closed 2019-08-07 by chrishamm