Duet3D/DuetSoftwareFramework #49: M905 at the end of config.g is executed on the system

wilriker, 2019-08-07:
```
duetcontrolserver=1.0.3.0-1
duetruntime=1.0.3.0-1
duetsd=1.0.1-3
duetsoftwareframework-meta=1.0.3.0-1
duettools=1.0.3.0-1
duetwebcontrol=2.0.0.RC8.b1-2
duetwebserver=1.0.1.1-2
duet3firmware=3.0b7@2019-08-05b3
```

With `DCS` 1.0.3.0 `M905` being sent at the end of `config.g` has been implemented. See #38.
I also found hints in the source code that this is supposed to then not try to set the SBC's date and time in this case but still it tries to. In the log of `DCS` and see
```
Aug 07 14:19:27 duet3-v05 DuetControlServer[14682]: Failed to set time: Automatic time synchronization is enabled
Aug 07 14:19:28 duet3-v05 DuetControlServer[14682]: Failed to set time: Automatic time synchronization is enabled
Aug 07 14:19:28 duet3-v05 DuetControlServer[14682]: [info] Sent M905 P"2019-08-07" S"14:19:27", remaining space 712 (1992 total), needed 56
```
where the first two lines of error messages are the output of `timedatectl set-time [...]` if time sync is enabled (as on my setup). This means the flag to execute this internally does not work as intended.

Also as a side-note: `M905` is automatically appended whenever **any** macro called `config.g` is executed, i.e. also when `M703` is executing a filament-specific `config.g`. It is not a problem but I want to still mention it.

chrishamm, 2019-08-07:
Thanks, actually it should be sent at the beginning of config.g. I'll fix the config.g detection in my next version as well (which I will hopefully release within the next hours).

chrishamm, 2019-08-07:
Fixed in DCS 1.0.3.1.

closed 2019-08-07 by chrishamm