Duet3D/DuetSoftwareFramework #53: M291 with jog buttons does not show current position

wilriker, 2019-08-09:
```
duetcontrolserver=1.0.3.1-1
duetruntime=1.0.3.1-1
duetsd=1.0.1-4
duettools=1.0.3.1-1
duetwebcontrol=2.0.0.RC8.b3-1
duetwebserver=1.1.0.0-1
duet3firmware=3.0b7-ch@2019-08-07b2
```

Now, this is most likely rather a DWC2 issue. In DWC1 if a `M291` dialog had jog buttons then the current position of the corresponding axis was shown in between negative and positive jog buttons. This is no longer the case in DWC2 and reduces the usefulness of this feature in some use-cases.

chrishamm, 2019-08-09:
Thanks for your reports concerning message boxes. I will rework the underlying mechanism in the next DSF version so every code can generate message boxes again.

chrishamm, 2019-12-19:
Axis positions are properly displayed in DWC 2.0.4.

closed 2019-12-19 by chrishamm