Duet3D/DuetSoftwareFramework #55: Auto save on loss of power not working

wilriker, 2019-08-15:
Currently saving the `resurrect.g` is bound to `HAS_MASS_STORAGE` and no alternative for `HAS_LINUX_INTERFACE` is provided. It will pause the print on power loss but it won't save resume information.

dc42, 2019-08-17:
Yes, we know about this. One of the problems is that the RPi is so power-hungry (+ the display if one is connected) that maintaining reserve power for long enough to save the resurrect data may be difficult anyway. If that can be overcome (perhaps by using a separate PSU for the RPi) then we're looking at reserving an area of flash memory in the Duet main processor to save the resurrect data. Or, if we decide to always populate the optional SD card socket on the v0.6 Duet main board, we can save it to SD card as before, and make the contents available to the Pi.

chrishamm, 2020-12-09:
Depends on #155 and #156.

ernstc, 2021-04-09:
Using an UPS like the [BTT UPS 24V](https://www.bigtree-tech.com/products/bigtreetech-ups-24v-v1-0.html) gives enough power to save the resurrect data. Since this components are available, it's better to have the feature of saving resurrect data available in some way, maybe with warnings, than not having at all.
Maybe a flag that can be enabled with a GCODE for saving data on SBC could work.

Tinchus2009, 2021-04-24:
Question: provided that duet 3 has enough power, would this feature work in SBC mode?