Duet3D/DuetSoftwareFramework #60: Thermistor Temperature Incorrect

techildllc, 2019-08-31:
Using any 100K thermistor directly on any of the thermistor inputs on the boards will yield a temperature ~20C below actual temperature (i.e. reading 5C when its 25C). This seems to be linear up to 90C. Measured using a K type thermocouple attached to Fluke DMM. 

RepRapFirmware for Duet 3 v0.5 version 3.0alpha running on Duet 3 prototype v0.5

dc42, 2019-08-31:
Please confirm what M308 line you are using. You should not have a R parameter in that line.

techildllc, 2019-08-31:
Ok, I just removed the R4700, and the value is now reading about the correct temperature vs the thermocouple. Was the R value the series resistor? Shouldnt that be 10K?

**Was:** M308 S0 P"temp0" Y"thermistor" T100000 B3950 C0 R4700
**Now:** M308 S0 P"temp0" Y"thermistor" T100000 B3950 C0

Now I see my values fluctuate a bit more at room temp, about ~1C. Wonder if thats just the ADC noise floor?

dc42, 2019-08-31:
Series resistors on Duet 2 are 2.2K. You never need to specify the R parameter, except when using a Duet 0.6 or 0.8.5.

techildllc, 2019-08-31:
Got it, perfect!

closed 2019-08-31 by techildllc