Duet3D/DuetSoftwareFramework #61: DWC jog step formatting fails if changing the sign

beikeland, 2019-09-11:
As using this for CNC I tried changing the Z jog direction so the left hand buttons would be towards the origin, and the right hand buttons away from the origin by entering a negative value for the right hand buttons. While this achieve the desired effect the formatting broke with showing both signs on the right side and no signs on the left side.

[image]

I guess DWC should reject negative values or fix the formatting.

chrishamm, 2019-10-08:
I'll fix negative jog values in DWC 2.0.1. Thanks for your report!

chrishamm, 2019-10-08:
Fixed in v2.0.1.

closed 2019-10-08 by chrishamm