Duet3D/DuetSoftwareFramework #73: Start duetwebserver and duetcontrolserver by default

natewalck, 2019-10-18:
Would it make sense to enable/start the duetwebserver and duetcontrolserver as part of the `postint` script in the package? Are there any downsides to do doing so?

chrishamm, 2019-10-21:
I am not sure how Raspbian handles this - AFAIK that is not the default behavior on Arch Linux but Raspbian/Debian may be different.

Do you know if it's common for Raspbian/Debian packages to start services after the (first) installation?

natewalck, 2019-10-21:
This is not typically the case. That said, I think most people who run Debian know what they are doing or ate using a config management tool. 

I was thinking that because of the demographic, it might be helpful to start it for them to reduce confusion.

Maybe a better idea here would be to make an interactive setup script that guides the user through initial config and starts the service. This script would have sane defaults, install the packages for the end user and then apply settings and start the services.

chrishamm, 2019-10-21:
Well, for those cases there will be a customized image that has all the required parts preconfigured including Avahi for mDNS (DuetPi, see my other repo). In that image, both DCS and DWS are started automatically.

natewalck, 2019-10-21:
Ah gotcha. Yea, then let’s leave this as is then. Doing it manually vs the official image should be discouraged for everyone except people who can figure it out imo. 

Also, is there a repo that has scripts for generating the custom Raspian distribution?Curious what all that entails.

chrishamm, 2019-10-21:
Yes, see https://github.com/chrishamm/pi-gen I'm about to generate a new version too.

closed 2019-10-21 by chrishamm