Duet3D/DuetSoftwareFramework #78: Two or more M122 in a macro crashes DSF

dc42, 2019-11-05:
Text below is from https://forum.duet3d.com/topic/12713/duet-3-two-or-more-m122-in-a-macro-crashes-firmware. I have verified that the macro runs OK in standalone mode.

Board: Duet 3 MB6HC (MB6HC)
DSF Version: 1.1.0.5
Firmware: RepRapFirmware for Duet 3 MB6HC v0.6 or 1.0 3.0beta12 (2019-11-02b1)

Macro containing:

M122 B0
M122 B1

or

M122 B0
M400
G4 S5
M400
M122 B1

Crashes on second M122. Re-ordering boards makes no difference. When it crashes, it does not reboot; it never comes back. Requires reset or power cycle. Log sometimes says:

Error: Operation failed (Reason: Object reference not set to an instance of an object.)

Each command works fine when entered separately from console.

chrishamm, 2019-11-05:
I cannot reproduce this with RRF3 beta11 but I'll update my board shortly and try it again.

chrishamm, 2019-11-06:
This was caused by a misconfigured module parameter.

closed 2019-11-06 by chrishamm