Duet3D/DuetSoftwareFramework #81: Build for LinuxDevices is creating DuetSoftwareFramework/src/LinuxDevices/C:/Users/...

gtjoseph, 2019-12-15:
The full path:
DuetSoftwareFramework/src/LinuxDevices/C:/Users/Christian/Source/repos/DuetSoftwareFramework/src/LinuxDevices

gtjoseph, 2019-12-15:
Oh, it's also creating pkg/package-lock.json

chrishamm, 2019-12-18:
The latest version of DSF doesn't generate any of those paths on my machines. Is there any way to reproduce this?

gtjoseph, 2019-12-18:
Linux vs Windows?
Happens every time for me just doing a plain...
```
$ ./build.sh --target-arch=armv7h --build-type=Release --packages=progs --no-pkgs deb
$ git status
On branch dev
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	../src/LinuxDevices/C:/

```

chrishamm, 2019-12-19:
Ah yes, it's generated every time the LinuxDevices library is built with the Release configuration. I'll fix it - thanks for report!

chrishamm, 2020-01-06:
This was fixed in DSF 1.2.1.

closed 2020-01-06 by chrishamm