Duet3D/DuetSoftwareFramework #83: M32 Print non functional in 1.2.00-rre

gtjoseph, 2019-12-15:
First issue...   When DCS starts _fileToPrint is uninitialized so it can't be used as a lock object.  This causes "System.ArgumentNullException: Value cannot be null"

After I fixed that, I started a print again and the start went to "processing" and stayed there.  Attempting to restart the DCS caused the printer to try and print "something" for a second or two, then hang again.  Had to restart both the DCS and Duet.

gtjoseph, 2019-12-15:
HA!  I had left the printer in the "processing" state while I opened this issue but 4 minutes after it hung, it suddenly started printing.

chrishamm, 2019-12-15:
I've changed the print processor to use an own task so I guess that will be fixed soon (not on GitHub yet). I hope to publish a new version tomorrow once I've finished a few test prints.

chrishamm, 2019-12-18:
Should be fixed in the latest version.

closed 2019-12-18 by chrishamm