Duet3D/DuetSoftwareFramework #84: Cancel after Pause executes moves left in the buffer

gtjoseph, 2019-12-17:
When you pause a print from sd card with M25, it takes a second or two to stop and execute pause.g.  That's fair.

If you then cancel the print with M0 H1, commands are being executed that are NOT in cancel.g nor stop.g.  From the movement, it appears that whatever commands are left in the buffer after the pause are now executed followed by stop.g.  Maybe a cancel should clear the buffers before running stop.g?

Actually this isn't a recent issue but I just now had time to try it a few times and observe the movements and timing.

chrishamm, 2019-12-17:
I can confirm this is a bug which will be fixed in the final 1.2 version.

chrishamm, 2019-12-18:
This is now fixed in DSF 1.2.

closed 2019-12-18 by chrishamm

wilriker, 2019-12-26:
Unfortunately I saw a remainder of this still in 1.2.1.0. If pausing was delayed I cannot say but then pressing Cancel in DWC the head first moved over the point where it was paused (only X and Y) and only then executed `stop.g`. It should have done nothing to at least the X axis since the final positions in both `pause.g` and `stop.g` are the same and there is only one move command in there. (Y is different but it clearly stopped for a tiny amount of time at the print's last Y position before moving on to final post-print park-position)

gtjoseph, 2019-12-30:
I'm also still seeing the issue in 1.2.1.0.  Actually seems to be worse for me.   I paused a print, then cancelled it.  I then cleaned the nozzle and restarted the print.   Unfortunately, the cancel was still "processing" and hadn't even done its moves yet.  Things got really confusing t=from there on. :)

gtjoseph, 2019-12-30:
And now I wonder why I can't re-open this issue.  Opening a new one.