Duet3D/DuetSoftwareFramework #86: Need a "Cancelling" status

gtjoseph, 2019-12-17:
There's already pausing, paused and resuming.  We shoul dhave a "Cancelling" state while executing the cancel sequence before going back to idle.

chrishamm, 2019-12-17:
I'll only implement this if it becomes part of RRF because that's where the machine status usually comes from. @dc42 what do you think?

chrishamm, 2021-07-12:
To get back to this: What's the motivation behind this new state? Most cancellation configurations complete almost instantly.

gtjoseph, 2021-07-12:
With a large printer, the cancel sequences can get fairly lengthy.  Not in the number of gcode commands but in the time it takes to execute them...  return to 0,0 from the far end of the build surface, drop the bed 300mm, etc.  It also just seemed to make "sense" since there was already a pausing and resuming state.

chrishamm, 2021-07-12:
Fair enough, I've just added this. Will discuss with David next week if we can integrate it into v3.4.