Duet3D/DuetSoftwareFramework #87: Add/Populate the machine model "job" object with missing data

gtjoseph, 2019-12-17:
We now have timesLeft.filament and timesLeft.file  (not for layer yet) but we don't know what their original values were so it's hard to calculate percentages complete.  We can't use file.printTime as the base because it's not accurate based on any thing I can find.  Maybe add the original values to the "file" object?

See below for other issues:
```
"job" : {
         "duration" : 27,
         "extrudedRaw" : [
            23.5
         ],
         "file" : {
            "filament" : [
               1196   (millimeters or time?)
            ],
            "fileName" : "0:/gcodes/Sample Retainer.gcode",
            "firstLayerHeight" : 0.2,
            "generatedBy" : "Simplify3D(R) Version 4.1.1",
            "height" : 0.95,
            "lastModified" : "2019-12-16T16:43:24.1369177-07:00",
            "layerHeight" : 0.25,
            "numLayers" : 4,
            "printTime" : 720,  (where was this calculated from?)
            "simulatedTime" : 0,
            "size" : 711504
         },
         "filePosition" : 17316,
         "lastFileAborted" : false,  (what's an abort vs cancel?)
         "lastFileCancelled" : false,
         "lastFileName" : "0:/gcodes/Sample Retainer.gcode",
         "lastFileSimulated" : false,
         "layer" : 1,
         "layerTime" : 27,
         "layers" : [],   (not pupulated)
         "timesLeft" : {
            "filament" : 1344.8,  (based on file.filament, extrudedRaw and duration?)
            "file" : 1081.6,   (based on file.filament, filePosition and duration?)
            "layer" : null  (not populated)
         },
         "warmUpDuration" : 0
      },
```

gtjoseph, 2019-12-17:
Oh, file.printTime was from the slicer. OK.

chrishamm, 2019-12-17:
What's the problem with leaving it in there? Many users requested the ability to see information about the last printed file. Perhaps it would make sense to add the print progress as a discrete field to the object model.

If a file is cancelled, thats' because of a M0/M1. If a severe error occurred, the file is aborted. You'll see the same behavior in RRF.

For any other details, please check out the new code documentation.

gtjoseph, 2019-12-17:
Yeah OK, I guess we can check the last(Aborted|Cancelled|Simulated) flags to suppress the display.  Any word on the missing stuff?

chrishamm, 2019-12-17:
Check out the docs [here](https://chrishamm.github.io/DuetSoftwareFramework/html/T_DuetAPI_ParsedFileInfo.htm) and [here](https://chrishamm.github.io/DuetSoftwareFramework/html/T_DuetAPI_Machine_Job.htm).

The new documentation is now on the `dev` branch but you need to clone it first to read it.

chrishamm, 2019-12-24:
The job progress is not part of any status response and it is usually calculated from the filament usage, see https://github.com/chrishamm/DuetWebControl/blob/master/src/store/machine/model.js#L245 That's why I am reluctant to add a new property to the OM. The fraction printed can be easily calculated from the file position and the file size,.

I know FilePosition wasn't properly updated in previous versions but I am going to fix this in v1.2.1.

chrishamm, 2020-01-02:
More fields are now populated from RRF/DCS when a print is being started/processed/stopped. Can you check DSF 1.2.2 again and let me know if you're missing anything else?

gtjoseph, 2020-01-02:
Will do!

chrishamm, 2020-01-02:
PS: I just had to update the RRF package again in order to fix a macro-related problem so make sure you're on the latest RRF version from my v3-chrishamm RRF branch. If everything goes well, we'll release DSF 1.2.2 and RRF 3.0 tomorrow.

gtjoseph, 2020-01-02:
Gotcha.  In progress.

gtjoseph, 2020-01-02:
Hmmm.  Where are the "correct" iap binaries now?  It's been a while since I've been able to update with a M997.  I always seem to need to use Bossac.

chrishamm, 2020-01-02:
I've just updated the latest RRF release, here the direct link: https://github.com/dc42/RepRapFirmware/releases/download/3.0RC2/Duet3_SBCiap_MB6HC.bin It will work on a MBP05, too.

Alternatively, you can build it from dc42's DuetIAP sources.

gtjoseph, 2020-01-02:
Yeah I tried building from the DuetIAP sources and it didn't work any better.  I'll use yours in the future.

gtjoseph, 2020-01-03:
Seems to be OK although the "layers" array still isn't populated.  I don't even know what's supposed to be in there but I don't miss it. :)

chrishamm, 2020-01-03:
Hmm, works on my setup. It's supposed to hold information about every printed layer like layer time, filament usage, and file progress.

closed 2020-01-06 by chrishamm