Duet3D/DuetSoftwareFramework #88: M26 not working

defux, 2019-12-17:
I am calling macro to start print with certain offset:

**M23 PrintTest.gcode
M26 S10070
M24**

Error I get:

**Error: Failed to execute M26: [NullReferenceException] Object reference not set to an instance of an object.**

Same thing runs as expected on non-beta setup of mine. Is there something else I am missing?

chrishamm, 2019-12-17:
Thanks, I can confirm this is a bug which will be fixed in the next release.

chrishamm, 2019-12-18:
M26 is now fixed in DSF 1.2 but the macro above may not be working yet. You may want to try out this instead (untested):

M32 "PrintTest.gcode"
M26 S10070

What's the reason you're trying to start a file print with an offset?

defux, 2019-12-18:
I worked on making it possible to "rewind" my print. Once portion of the print was re-winded without extrusion and paused, I'd run macro "continue Original Print from here". Initially I wanted to use M26 so I always run same file, just with different offset, but then I noticed M26 doesn't work. I made a workaround, however I am pretty sure it would be more elegant with M26.

On somewhat related matter, "Resurrect.g" is not being generated on pause. Is it a known issue or should I make a separate post?

chrishamm, 2019-12-18:
I see. In that case you may be better off by running

M226
M26 S10070

when your print has finished - either directly or in a macro file. I'll change DSF 1.2.1 to support dynamic start positions though. Regarding resurrect.g, see #55.

defux, 2019-12-18:
Thanks for the tip

chrishamm, 2020-01-02:
Selectng a print, setting a byte offset, and resuming a print from there on should be possible with DSF 1.2.2 and later.

closed 2020-01-02 by chrishamm