Duet3D/DuetSoftwareFramework #89: Machine model incorrectly maps drives to axes/extruders

gtjoseph, 2019-12-17:
With the following in config.g:
```
M584 X0 Y1 E2 Z3:4:5
M906 X1750 Y1750 E1500 Z1750
```

I get this: (issues marked with ********)
```
"move" : {
         "axes" : [
            {
               "drives" : [
                  0
               ],
               "homed" : true,
               "letter" : "X",
               "machinePosition" : 223.718,
               "max" : 460,
               "maxEndstop" : 0,
               "maxProbed" : false,
               "min" : 0,
               "minEndstop" : 0,
               "minProbed" : false,
               "visible" : true
            },
            {
               "drives" : [
                  1
               ],
               "homed" : true,
               "letter" : "Y",
               "machinePosition" : 195.327,
               "max" : 450,
               "maxEndstop" : 1,
               "maxProbed" : false,
               "min" : 0,
               "minEndstop" : 1,
               "minProbed" : false,
               "visible" : true
            },
            {
               "drives" : [
                  2   ******** should be 3,4,5
               ],
               "homed" : true,
               "letter" : "Z",
               "machinePosition" : 15.055,
               "max" : 465,
               "maxEndstop" : 2,
               "maxProbed" : false,
               "min" : 0,
               "minEndstop" : 2,
               "minProbed" : false,
               "visible" : true
            }
         ],
         "drives" : [
            {
               "acceleration" : 3600,
               "current" : 1750,
               "maxSpeed" : 266.67,
               "microstepping" : {
                  "interpolated" : true,
                  "value" : 16
               },
               "minSpeed" : 10,
               "position" : 223.718
            },
            {
               "acceleration" : 3600,
               "current" : 1750,
               "maxSpeed" : 266.67,
               "microstepping" : {
                  "interpolated" : true,
                  "value" : 16
               },
               "minSpeed" : 10,
               "position" : 195.327
            },
            {    ******** this should be the extruder, not Z
               "acceleration" : 125,
               "current" : 1750,
               "maxSpeed" : 20,
               "microstepping" : {
                  "interpolated" : true,
                  "value" : 16
               },
               "minSpeed" : 10,
               "position" : 15.055
            },
            {     ******** this should be the first Z drive
               "acceleration" : 500,
               "current" : 0,
               "maxSpeed" : 100,
               "microstepping" : {
                  "interpolated" : true,
                  "value" : 16
               },
               "minSpeed" : 15,
               "position" : 6.6
            },
                 ******** the second and third Z drives should be here but they're empty
                 ******** snipped 15 unconfigured drives
           {    ******** This is the extruder drive but it should be in the number 2 spot
               "acceleration" : 6000,
               "current" : 1500,
               "maxSpeed" : 100,
               "microstepping" : {
                  "interpolated" : true,
                  "value" : 16
               },
               "minSpeed" : 10,
               "position" : null
            }
         ],
         "extruders" : [
            {
               "drives" : [
                  3    ********  This should be drive 2
               ],
               "factor" : 1,
               "nonlinear" : {
                  "a" : 0,
                  "b" : 0,
                  "temperature" : 0,
                  "upperLimit" : 0.2
               }
            }
         ],
```

chrishamm, 2019-12-17:
Closing again, see #69. There is nothing I can do without RRF returning the right mapping.
If you check out the Updater code, you'll see that it only tries to reconstruct the drives from the information it receives.

closed 2019-12-17 by chrishamm

gtjoseph, 2019-12-17:
So how are we supposed to know whether an issue is DSF or RRF?????