Duet3D/DuetSoftwareFramework #90: M500 NullReference in 1.2.0.0 if no "P" parameter is supplied

gtjoseph, 2019-12-18:
```
[error] Code M500 has caused an exception
 System.NullReferenceException: Object reference not set to an instance of an object.
 at DuetAPI.Commands.CodeParameter.op_Implicit(CodeParameter codeParameter) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Commands/Code/Parameter.cs:line 364
at DuetControlServer.Utility.ConfigOverride.Save(Code code) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/Utility/ConfigOverride.cs:line 23
```
If you supply any P parameter, even if invalid, the command will complete successfully.

chrishamm, 2019-12-19:
Thanks for the note, I'll fix it in DSF 1.2.1.

chrishamm, 2020-01-02:
Fixed in DSF 1.2.2.

closed 2020-01-02 by chrishamm