Duet3D/DuetSoftwareFramework #93: Still having issues with Cancel, pause and Resume in 1.2.1.0

gtjoseph, 2019-12-30:
And so is @wilriker ...

I'm still seeing the issue in 1.2.1.0. Actually seems to be worse for me. I paused a print, then cancelled it. I then cleaned the nozzle and restarted the print. Unfortunately, the cancel was still "processing" and hadn't even done its moves yet. Things got really confusing from there on. :)

See latest comments in #84

chrishamm, 2020-01-02:
Should be fixed in DSF 1.2.2.

closed 2020-01-02 by chrishamm

gtjoseph, 2020-01-03:
Pause is no longer executing moves from the buffer but Cancel still is.

wilriker, 2020-01-03:
To be clear: for me the issue in 1.2.1.0 only was with Cancel and then only a single XY move back over the object (where X should not have moved and Y simply front to back). I have not yet had the time to check with 1.2.2.0.