Duet3D/DuetSoftwareFramework #94: Still having issues with Cancel in DSF 1.2.2.0

gtjoseph, 2020-01-03:
Pause is no longer executing moves from the buffer but Cancel still is.

You might want to leave this one open until I can confirm it's actually fixed. :)

chrishamm, 2020-01-03:
Can you give me an example for this? I can pause/cancel a print and no extra codes from the buffer are executed. Also, the code in DSF makes sure pending codes are cancelled.

There is one thing I've just spotted and it's a missing error message in DSF in case a user tries to cancel a print without pausing it first (unless M0/1 is called from the file itself) so I'll fix that in the next version.

gtjoseph, 2020-01-03:
I think that noticing it depends on the code that's left in the buffer, if you're printing a radius with tons of segments, you probably won't notice the extra move because it's so tiny.  If you're printing something that has long moves it's much more obvious.  Try printing a large square and see what happens.  In the mean time, I'll up the log level on the DCS and see if I can capture the extra codes.

chrishamm, 2020-01-06:
I'm sorry but I have not been able to provoke this behavior once. Perhaps it was somehow related to the out-of-order execution you reported in another ticket but I took the time to review everything related to cancelling in DSF and RRF and yet I could not find one potential place that could explain this. Of course I made a new G-code file with long moves as well and tried to reproduce it by pausing/resuming/cancelling that file various times but it always worked as expected on my setup.

Please check if DSF 1.2.3 solves this problem and if it does not, please provide accurate instructions on how this can be reproduced.

gtjoseph, 2020-01-07:
I meant to create a test gcode file over the weekend but didn't get to it.  I'll do it on 1.2.2.0 first then try 1.2.3.

gtjoseph, 2020-01-07:
Still having the issue with 1.2.3.0.  Here's some output from the DCS. Not sure if that warning is significant or not.

```
[debug] Cancelled G1 X35.642 Y61.384
[info] Executing nested macro file pause.g on channel HTTP
[debug] ==> Starting code: M25
[debug] Waiting for execution of ; pause.g (type 1)
[debug] Processing ; pause.g
[debug] Completed ; pause.g
[debug] Waiting for execution of ; called when a print from SD card is paused (type 1)
[debug] Processing ; called when a print from SD card is paused
[debug] Completed ; called when a print from SD card is paused
[debug] Waiting for execution of ;M83            ; relative extruder moves (type 1)
[debug] Processing ;M83            ; relative extruder moves
[debug] Completed ;M83            ; relative extruder moves
[debug] Waiting for execution of ;G1 E-5 F600   ; retract 5mm of filament (type 1)
[debug] Processing ;G1 E-5 F600   ; retract 5mm of filament
[debug] Completed ;G1 E-5 F600   ; retract 5mm of filament
[debug] Waiting for execution of G91; relative positioning (type 1)
[debug] Processing G91; relative positioning
[debug] Waiting for execution of G1 Z5 F600; lift Z by 5mm (type 1)
[debug] Processing G1 Z5 F600; lift Z by 5mm
[debug] HTTP: Sent G91; relative positioning, remaining space 1516, needed 20
[debug] HTTP: Sent G1 Z5 F600; lift Z by 5mm, remaining space 1480, needed 36
[debug] HTTP: Finished macro file pause.g
[debug] Completed G1 Z5 F600; lift Z by 5mm
[debug] Completed G91; relative positioning
[info] HTTP: Finished macro file pause.g
[debug] Completed M25 => Printing paused at X39.4 Y83.6 Z10.0
[debug] HTTP: ==> Starting code: M25 => Printing paused at X39.4 Y83.6 Z10.0
[debug] IPC#26: Connection closed
[debug] IPC#27: Got new UNIX connection, checking mode...
[debug] IPC#27: Command processor added
[debug] IPC#27: Received command SimpleCode
[debug] Waiting for execution of M0 H1
[debug] Processing M0 H1
[info] Cancelled print file
[debug] HTTP: Sent M0 H1, remaining space 1508, needed 28
[info] Cancelled printing file 0:/gcodes/sr.gcode, print time was 0h 0m
[warn] File: Out-of-order reply: ''
[info] Executing nested macro file stop.g on channel HTTP
[debug] ==> Starting code: M0 H1
[debug] Waiting for execution of ; stop.g (type 1)
[debug] Processing ; stop.g
[debug] Completed ; stop.g
[debug] Waiting for execution of ; called when M0 (Stop) is run (e.g. when a print from SD card is cancelled) (type 1)
[debug] Processing ; called when M0 (Stop) is run (e.g. when a print from SD card is cancelled)
[debug] Completed ; called when M0 (Stop) is run (e.g. when a print from SD card is cancelled)
[debug] Waiting for execution of ;M83 (type 1)
[debug] Processing ;M83
[debug] Completed ;M83
[debug] Waiting for execution of ;G1 E-5 F100 (type 1)
[debug] Processing ;G1 E-5 F100
[debug] Completed ;G1 E-5 F100
[debug] Waiting for execution of G91; relative positioning (type 1)
[debug] Processing G91; relative positioning
[debug] Waiting for execution of G1 Z5 F600; lift Z by 5mm (type 1)
[debug] HTTP: Sent G91; relative positioning, remaining space 1516, needed 20
[debug] Processing G1 Z5 F600; lift Z by 5mm
[debug] HTTP: Sent G1 Z5 F600; lift Z by 5mm, remaining space 1480, needed 36
[debug] Completed G91; relative positioning
[debug] HTTP: Finished macro file stop.g
[debug] Completed G1 Z5 F600; lift Z by 5mm
[info] HTTP: Finished macro file stop.g
[debug] Completed M0 H1
[debug] HTTP: ==> Starting code: M0 H1
```
And here's the gcode file I was printing.  It's a stripped version of a real print file but without any extrusion or heat, etc. so it's safe to run.   It just prints some rounded corner rectangles.  Start it printing and give it about 15-20 seconds and pause with just an `M25`, then wait a few seconds and cancel with an `M0 H1`.  At the very least, you should see the XY position display change and if the pause is executed as the printer is about to print one of the sides, you'll see the carriage change actual position.

[sr.gcode.txt](https://github.com/chrishamm/DuetSoftwareFramework/files/4031311/sr.gcode.txt)

My stop.g and pause.g are simple...
```
G91            ; relative positioning
G1 Z5 F600     ; lift Z by 5mm
```

chrishamm, 2020-01-07:
Thanks for the feedback but I'm afraid the log does not really explain why you are encountering this issue. There are no more File codes (in whatever order) sent to the printer when M25 is called. Also I cannot spot any out-of-order execution of the macro files.

Can you please flash the latest RRF3 firmware binary from dc42's repository and enable debugging for the GCodes module via `M111 P3 S1` if this problem persists? There was definitely an issue with out-of-order replies in previous firmware versions but AFAICT it has been fixed in the latest one.

The only thing I find surprising is that there is no "Print has been paused at byte xxx" in the log. Was it perhaps somewhere else in your log? At the time this message is printed, RRF's internal code buffer is invalidated and pending codes are thrown away.

gtjoseph, 2020-01-08:
Sorry...  Here's what was leading up to the log I pasted above...
```
[debug] Waiting for execution of G1 X35.642 Y61.384
[debug] Processing G1 X35.642 Y61.384
[debug] File: Sent G1 X37.343 Y77.996, remaining space 852, needed 36
[debug] File: Sent G1 X36.859 Y77.195, remaining space 816, needed 36
[debug] File: Sent G1 X36.772 Y77.036, remaining space 780, needed 36
[debug] File: Sent G1 X36.534 Y76.547, remaining space 744, needed 36
[debug] File: Sent G1 X36.462 Y76.380, remaining space 708, needed 36
[debug] File: Sent G1 X36.141 Y75.534, remaining space 672, needed 36
[debug] File: Sent G1 X36.084 Y75.361, remaining space 636, needed 36
[debug] File: Sent G1 X35.938 Y74.837, remaining space 600, needed 36
[debug] File: Sent G1 X35.898 Y74.660, remaining space 564, needed 36
[debug] File: Sent G1 X35.729 Y73.741, remaining space 528, needed 36
[debug] File: Sent G1 X35.700 Y73.501, remaining space 492, needed 36
[debug] File: Sent G1 X35.645 Y72.596, remaining space 456, needed 36
[debug] File: Sent G1 X35.642 Y72.476, remaining space 420, needed 36
[debug] File: Sent G1 X35.642 Y61.384, remaining space 384, needed 36
[debug] IPC#26: Got new UNIX connection, checking mode...
[debug] IPC#26: Command processor added
[debug] IPC#26: Received command SimpleCode
[debug] Waiting for execution of M25
[debug] Processing M25
[debug] HTTP: Sent M25, remaining space 1516, needed 20
[info] Print paused at file position 3511. Reason: User
[debug] Print has been paused at byte 3511
[debug] Cancelled G1 X40.365 Y82.920
[debug] Cancelled G1 X41.401 Y82.360
[debug] Cancelled G1 X41.565 Y82.283
[debug] Cancelled G1 X42.067 Y82.075
[debug] Cancelled G1 X42.237 Y82.013
[debug] Cancelled G1 X42.537 Y81.919
[debug] Cancelled G1 X42.124 Y81.777
[debug] Cancelled G1 X41.955 Y81.710
[debug] Cancelled G1 X41.101 Y81.326
[debug] Cancelled G1 X40.887 Y81.213
[debug] Cancelled G1 X40.571 Y82.795
[debug] Cancelled G1 X40.061 Y80.714
[debug] Cancelled G1 X39.101 Y79.980
[debug] Cancelled G1 X38.920 Y79.820
[debug] Cancelled G1 X38.237 Y79.137
[debug] Cancelled G1 X38.076 Y78.956
[debug] Cancelled G1 X37.480 Y78.195
[debug] Cancelled G1 X37.343 Y77.996
[debug] Cancelled G1 X36.859 Y77.195
[debug] Cancelled G1 X36.772 Y77.036
[debug] Cancelled G1 X36.534 Y76.547
[debug] Cancelled G1 X39.862 Y80.576
[debug] Cancelled G1 X36.462 Y76.380
[debug] Cancelled G1 X36.141 Y75.534
[debug] Cancelled G1 X36.084 Y75.361
[debug] Cancelled G1 X35.938 Y74.837
[debug] Cancelled G1 X35.898 Y74.660
[debug] Cancelled G1 X35.729 Y73.741
[debug] Cancelled G1 X35.645 Y72.596
[debug] Cancelled G1 X35.700 Y73.501
[debug] Cancelled G1 X35.642 Y72.476
[debug] Cancelled G1 X35.642 Y61.384
[info] Executing nested macro file pause.g on channel HTTP
```

Still have the issue with @dc42 's 3.0 pre-built binary.

Attempting to print with `M111 P3 S1` enabled results in the Duet3 hanging and needint a reset.

```
M122
=== Diagnostics ===
RepRapFirmware for Duet 3 MB6HC v0.6 or 1.0 version 3.0 running on Duet 3 MB6HC
Board ID: 08DGM-9T66A-G63SJ-6J9F4-3SD6S-1U03B
[... 49 lines cut ...]
autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Slowest loop: 0.46ms; fastest: 0.01ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0) Telnet(0)
HTTP sessions: 0 of 8
- Ethernet -
State: 0
Error counts: 0 0 0 0 0
Socket states: 0 0 0 0 0 0 0 0
=== CAN ===
Messages sent 53, longest wait 0ms for type 0
=== Linux interface ===
State: 0, failed transfers: 0
Last transfer: 16ms ago
RX/TX seq numbers: 9637/446
SPI underruns 0, overruns 0
Number of disconnects: 0
Buffer RX/TX: 0/0-0
=== Duet Control Server ===
Duet Control Server v1.2.3.0
Code buffer space: 4096
Configured SPI speed: 2000000 Hz
Full transfers per second: 16.84
```

chrishamm, 2020-01-08:
Yes, that's all looking good. DCS stops sending instructions to the firmware when it is supposed to and RRF throws away the pending codes at the right time. At present I have no explanation why you are seeing additional movement when the print is cancelled.

You need to hook up the Pi or a PC to the USB port of the Duet 3 and record the USB log once debugging via M111 has been enabled. If you are using DuetPi, you only need to start minicom.

Can you add M122 to pause.g and send me the output, too?

gtjoseph, 2020-01-08:
M122 in pause.g displays nothing however, I've got more info for you...

I did an `M111 P3 S1` from the USB console and started the print, then did a pause

```
file: G1 X41.401001 Y82.360001
file: G1 X41.564999 Y82.282997
file: G1 X42.067001 Y82.074997
file: G1 X42.237000 Y82.013000
file: G1 X42.536999 Y81.918999 *******
file: G1 X42.124001 Y81.777000
file: G1 X41.955002 Y81.709999
file: G1 X41.101002 Y81.325996
file: G1 X40.887001 Y81.212997
file: G1 X40.061001 Y80.713997
file: G1 X39.862000 Y80.575996
file: G1 X39.101002 Y79.980003
file: G1 X38.919998 Y79.820000
file: G1 X38.237000 Y79.137001
file: G1 X38.076000 Y78.956001
file: G1 X37.480000 Y78.195000
file: G1 X37.342999 Y77.996002
file: G1 X36.859001 Y77.195000
file: G1 X36.771999 Y77.036003
file: G1 X36.534000 Y76.546997
file: G1 X36.462002 Y76.379997
file: G1 X36.140999 Y75.533997
file: G1 X36.084000 Y75.361000
file: G1 X35.938000 Y74.836998
file: G1 X35.897999 Y74.660004
file: G1 X35.729000 Y73.740997
file: G1 X35.700001 Y73.500999
file: G1 X35.645000 Y72.596001
file: G1 X35.641998 Y72.475998
file: G1 X35.641998 Y61.383999
file: G1 X35.641998 Y61.324001
file: G1 X35.657001 Y60.841999
file: G1 X35.671001 Y60.660999
file: G1 X35.784000 Y59.729000
file: G1 X35.827999 Y59.487000
file: G1 X36.066002 Y58.530998
file: G1 X36.138000 Y58.303001
file: G1 X36.473999 Y57.417999
file: G1 X36.573002 Y57.199001
file: G1 X37.021000 Y56.345001
file: G1 X37.146000 Y56.138000
file: G1 X37.695000 Y55.342999
file: G1 X37.844002 Y55.153000
file: G1 X38.485001 Y54.429001
file: G1 X38.655998 Y54.257999
file: G1 X39.379002 Y53.618000
file: G1 X39.854000 Y53.855999 F7200
http: M25
Paused print, file offset=3663
http: G91
http: G1 Z5 F600
http: G90
http: M122
```

At this time, the displayed coordinates were X42.537 Y81.919 which do NOT match the last move reported before the M25, nor the physical location of the toolhead.  In fact, X42.537 Y81.919 is about 44 moves _back_.

I then sent a cancel.
The toolhead and the display both moved to X39.854 Y53.855 to match the last reported move above.  Then...

```
Done printing file
Cancelled printing file 0:/gcodes/sr.gcode, print time was 0h 0m
http: M0 H1
http: G91
http: G1 Z5 F600
http: G90
http: M122
```

chrishamm, 2020-01-08:
AFAIR RRF does not report the current head position but the position of the last read move. When a print is paused, the move engine looks for the first move where the print can be paused and then reports/rewinds back to the file position after this move. That's what you see when you get the "Print paused at byte xxx" message.

In the logs above I still cannot see any direct fault of DSF - no new move from File is processed after M25 or M0 H1. So I suspect this is rather related to RepRapFirmware. However I won't be able to run further tests before next week. Does this problem still show up when you clear stop.g?

chrishamm, 2020-01-08:
Thinking further about it, my best guess is that the move buffer isn't properly reset at the time M0/stop.g is called. I'll try to figure out if that is the case.

gtjoseph, 2020-01-08:
Yes, with an empty stop.g, the rest of the moves between the reported position and the last "file:" debug statement are executed on cancel.

dc42, 2020-01-09:
@gtjoseph,  please can you clarify exactly what moves get executed on cancel. In an earlier message you said "I then sent a cancel. The toolhead and the display both moved to X39.854 Y53.855 to match the last reported move above." That makes it sound that only a single move was executed, not the whole sequence of skipped moves.

gtjoseph, 2020-01-09:
It wasn't apparent before but I verified that it's the "rest" of the moves in sequence.  So for instance, with my rounded corner rectangles above, when I pressed pause, the toolhead stopped just before rounding a corner which was the `file: G1 X42.536999 Y81.918999 *******` line.  Those coordinates were what was displayed.  When I pressed cancel, the toolhead moved around the corner and stopped at the coordinates stated in the last line in the debug `file: G1 X39.854000 Y53.855999 F7200`.

wilriker, 2020-01-12:
I can chime in here again because I can also see this behavior still in 1.2.3.0.

I was printing a hollow single-walled cube all day repeatedly (testing PA and DAA settings). It has the layer change front center. I paused it when it reached the back right corner, it executed `pause.g` just fine. Then when cancelling I first expected it to see it returning to where it left off. Instead it returned to the layer seam and then continued with `stop.g`.

Looking at DCS log
```
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [info] Print paused at file position 12452. Reason: User
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Print has been paused at byte 12452
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G0 F12000 X110 Y60.225 Z8.9
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 F2400 X159.775 Y60.225 E1.86246
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X159.775 Y159.775 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X60.225 Y159.775 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X60.225 Y60.225 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X110 Y60.225 E1.86246
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G0 F12000 X110 Y60.225 Z9.1
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 F2400 X159.775 Y60.225 E1.86246
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X159.775 Y159.775 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X60.225 Y159.775 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X60.225 Y60.225 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X110 Y60.225 E1.86246
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G0 F12000 X110 Y60.225 Z9.3
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled M572 D0 S0.1
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 F2400 X159.775 Y60.225 E1.86246
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X159.775 Y159.775 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X60.225 Y159.775 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X60.225 Y60.225 E3.72493
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G1 X110 Y60.225 E1.86246
Jan 12 17:34:56 duet3-6HC DuetControlServer[9180]: [debug] Cancelled G0 F12000 X110 Y60.225 Z9.5
```
it's the first cancelled command that it then executed although being cancelled because `X110 Y60.225 Z8.9` is where it returned to.

I have not checked what DWC showed at that point in time.

chrishamm, 2020-01-12:
Can you provide both the DCS and M111 P3 S1 logs? I doubt the G0 code is actually fed to RRF because a cancelled code cannot be sent to RRF since it can be only cancelled before it is processed. The logs should confirm this and hopefully reveal the actual cause.

wilriker, 2020-01-12:
Here you go: `M111 P3 S1` output:
```
$ miniterm.py --echo /dev/ttyACM0 115200                                                                                                                 Sun 12 Jan 2020 17:58:00 CET
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

Error: Bad command: Connection to Linux
ok
M111 P3 S1
Debugging enabled for modules: GCodes(3)
Debugging disabled for modules: Platform(0) Network(1) Webserver(2) Move(4) Heat(5) DDA(6) Roland(7) Scanner(8) PrintMonitor(9) Storage(10) PortControl(11) DuetExpansion(12) Filamen
tSensors(13) WiFi(14) Display(15) LinuxInterface(16)
http: M80
file: M80
file: T0
file: G21
file: G90
file: M83
file: M106 P0 S0
file: M703
file: M572 D0 S0.050000
file: M591 D0 R85:107 S1
file: M140 S60
file: M116 H0
http: G10 P0 S205
file: G10 P0 S205
file: M116 P0
http: M220 S125
file: M140 S55
file: G28
file: M80
file: G4 P750
file: G91
file: G1 H2 Z2 F3000
file: M400
file: M913 X40 Y45
file: G1 H1 X-300 Y-300 F3600
file: G1 X5 Y5 F3000
file: G1 H1 X-300 Y-300 F360
file: M98 P"homez.g"
file: M80
file: G4 P750
file: G91
file: M400
file: M913 Z50
file: G1 H1 Z-300 F900
file: G1 Z1 F3000
file: G1 H1 Z-300 F180
file: M584 P4
file: G1 H2 W0.340000
file: M584 P3
file: M400
file: G0 Z0.500000
file: M913 Z100
file: G90
file: G92 Z0
file: M400
file: M913 X100 Y100
file: G90
file: G0 Z0 F600
file: G0 Z2
file: M106 P2 S0.550000
file: T0
queue: M106 P2 S0.550000
file: M82
file: G11
file: M83
file: G10
file: M107
file: G0 F6000 X57.764999 Y57.693001 Z0.300000
[... 1084 lines cut ...]
Jan 12 18:02:39 duet3-6HC DuetControlServer[716]: [debug] Completed G1 Z+4 X-20 Y+20 F9000; move nozzle away from object
Jan 12 18:02:39 duet3-6HC DuetControlServer[716]: [debug] Completed G0 X-300 F3600; move X to the left, so the head is completely out of the way
Jan 12 18:02:39 duet3-6HC DuetControlServer[716]: [debug] Completed G90; absolute positioning
Jan 12 18:02:39 duet3-6HC DuetControlServer[716]: [debug] Completed G0 Y160 F5000; move bed to the front
Jan 12 18:02:39 duet3-6HC DuetControlServer[716]: [debug] Completed M591 D0; output stats on LFS => Duet3D laser filament monitor v1 on pin io5.in, enabled, allow 85% to 107%, check
 every 3.0mm, version 1, measured min 92% avg 95% max 98% over 90.3mm
Jan 12 18:02:39 duet3-6HC DuetControlServer[716]: [info] Duet3D laser filament monitor v1 on pin io5.in, enabled, allow 85% to 107%, check every 3.0mm, version 1, measured min 92% a
vg 95% max 98% over 90.3mm
Jan 12 18:02:39 duet3-6HC DuetControlServer[716]: [debug] Completed M591 D0 S0; Disable Laser Filament Sensor
Jan 12 18:02:42 duet3-6HC DuetControlServer[716]: [debug] Completed M84; steppers off
Jan 12 18:02:42 duet3-6HC DuetControlServer[716]: [debug] HTTP: Finished macro file stop.g
Jan 12 18:02:42 duet3-6HC DuetControlServer[716]: [debug] Completed M106 P2 S0; Stop hardware cooling
Jan 12 18:02:42 duet3-6HC DuetControlServer[716]: [debug] Completed M81; Turn off VIN PSU - fans run on 5V and are not affected
Jan 12 18:02:42 duet3-6HC DuetControlServer[716]: [info] HTTP: Finished macro file stop.g
Jan 12 18:02:42 duet3-6HC DuetControlServer[716]: [debug] Completed M0 H1
Jan 12 18:02:42 duet3-6HC DuetControlServer[716]: [debug] HTTP: ==> Starting code: M0 H1
Jan 12 18:02:44 duet3-6HC DuetControlServer[716]: [warn] 12V under-voltage event (9.5V)
```
I paused the print when it reached `X110 Y60.224998` and when cancelling it returned to `X110 Y60.224998 Z1.300000`.

From the logs you can see that DCS states it
```
Cancelled G0 F12000 X110 Y60.225 Z1.3
```
but in `M111` log you can see it reached it's target - though it was only executed when cancelling the paused print.

chrishamm, 2020-01-12:
Ah yes, the SPI connector invalidates the codes being processed by RRF when it receives a notification about the print being paused. Hence it's cancelled after M25. However the M111 log clearly shows that no new codes are being fed/interpreted by RRF once M25 is processed, so the error must be somewhere in RepRapFirmware.

I'll try to debug this in a few days unless @dc42 beats me to it.

dc42, 2020-01-13:
I've tried to reproduce this using the latest RRF3 build, and failed. I am using this file:

G1 F300

G1 X0.2
G1 Y0.2
G1 X0
G1 Y0

G1 X0.2
G1 Y0.2
G1 X0
G1 Y0
(repeat the above 4 lines many times)

I tested on Duet 3 in both standalone mode and with the RPi. No moves get executed after cancel. The file is designed so that the pause can be done almost immediately, because the feed rate is lower than the configured XY jerk. Also the moves are short so that the move buffer should be full, in case the problem only happens when the move buffer is full.

@gtjoseph, please can you test this file on your machine. If it doesn't fail with this file, please provide a file that it does fail on.

dc42, 2020-01-13:
With @wilriker's help, I was able to reproduce this, although I only ever saw a single command executed after cancelling the print. I believe this is fixed in RRF 3.1beta 2020-01-13b3.

wilriker, 2020-01-14:
I can confirm this fixed with the latest 3.01beta. I would say if @gtjoseph can also confirm this one could be closed.

gtjoseph, 2020-01-14:
Yep, just tried it with the current 3.01-dev and it's fixed.  Thanks!

closed 2020-01-14 by gtjoseph