Duet3D/DuetSoftwareFramework #95: Codes executed out of order in homing files (and possibly other macros)

gtjoseph, 2020-01-03:
Here's my homez.g:
```
; homez.g
G30
G1 Z0
G29 S1 P"heightmap.csv"
```

And here's the trace from the DCS.  Notice that the G29 _appears_ to finish before everything else.

```
[debug] IPC#28: Received command SimpleCode
[debug] Waiting for execution of G28 Z
[debug] Processing G28 Z
[debug] HTTP: Sent G28 Z, remaining space 1508, needed 28
[info] Executing nested macro file homez.g on channel HTTP
[debug] => Starting code: G28 Z
[debug] Processing G29 S1 P"heightmap.csv"
[debug] Processing G1 Z0
[debug] Processing G30
[debug] Processing ; homez.g
[debug] Completed ; homez.g
[debug] HTTP: Sent G30, remaining space 1516, needed 20
[debug] HTTP: Sent G1 Z0, remaining space 1488, needed 28
[info] Executing nested macro file deployprobe.g on channel HTTP
[debug] => Starting code: G30
[debug] HTTP: Suspending code G1 Z0
[debug] Completed G29 S1 P"heightmap.csv" => Height map loaded from file heightmap.csv
[info] Height map loaded from file heightmap.csv
[debug] HTTP: Finished macro file deployprobe.g
[debug] HTTP: Sent G1 Z0, remaining space 1508, needed 28
[debug] HTTP: -> Resumed suspended code
[info] HTTP: Finished intermediate macro file deployprobe.g
[info] Executing nested macro file retractprobe.g on channel HTTP
[debug] => Starting code: G30
[debug] HTTP: Suspending code G1 Z0
[debug] HTTP: Finished macro file retractprobe.g
[debug] HTTP: Sent G1 Z0, remaining space 1508, needed 28
[debug] HTTP: -> Resumed suspended code
[info] HTTP: Finished macro file retractprobe.g
[debug] HTTP: => Starting code: G30
[debug] HTTP: Finished macro file homez.g
[info] HTTP: Finished macro file homez.g
[debug] Completed G28 Z
[debug] Completed G1 Z0
[debug] Completed G30
[debug] HTTP: => Starting code: G28 Z
```

chrishamm, 2020-01-04:
Thanks for the report, I've got an idea why it may occur. I'll provide a fix soon.

chrishamm, 2020-01-06:
The issue with G29 was particularly interesting because it revealed another bug in the SPI subsystem. Nevertheless, I did various tests today and I was not able to create an out-of-order execution once with the improvements I made to the code scheduler.

I also tested your homing macro and it seems to be working well on my setup with v1.2.3.

gtjoseph, 2020-01-07:
Cool.  I'll test it in the morning.

gtjoseph, 2020-01-07:
No change with 1.2.3.0.  Still seeing the completion message for the heightmap before the execution of the movement gcodes, even with an M400 after the last movement command.

With
```
; homez.g
G30
G1 Z0
M400
G29 S1 P"heightmap.csv"
```
I get...

```
[debug] IPC#32: Got new UNIX connection, checking mode...
[debug] IPC#32: Command processor added
[debug] IPC#32: Received command SimpleCode
[debug] Waiting for execution of G28 Z
[debug] Processing G28 Z
[debug] HTTP: Sent G28 Z, remaining space 1508, needed 28
[info] Executing nested macro file homez.g on channel HTTP
[debug] ==> Starting code: G28 Z
[debug] Waiting for execution of ; homez.g (type 1)
[debug] Processing ; homez.g
[debug] Completed ; homez.g
[debug] Waiting for execution of G30 (type 1)
[debug] Processing G30
[debug] Waiting for execution of G1 Z0 (type 1)
[debug] Processing G1 Z0
[debug] Waiting for execution of M400 (type 1)
[debug] Processing M400
[debug] Waiting for execution of G29 S1 P"heightmap.csv" (type 1)
[debug] Processing G29 S1 P"heightmap.csv"
[debug] HTTP: Sent G30, remaining space 1516, needed 20
[debug] HTTP: Sent G1 Z0, remaining space 1488, needed 28
[debug] HTTP: Sent M400, remaining space 1468, needed 20
[info] Executing nested macro file deployprobe.g on channel HTTP
[debug] ==> Starting code: G30
[debug] HTTP: Suspending code G1 Z0
[debug] HTTP: Suspending code M400
[debug] Completed G29 S1 P"heightmap.csv" => Height map loaded from file heightmap.csv
[info] Height map loaded from file heightmap.csv
[debug] HTTP: Finished macro file deployprobe.g
[debug] HTTP: Sent G1 Z0, remaining space 1508, needed 28
[debug] HTTP: -> Resumed suspended code
[debug] HTTP: Sent M400, remaining space 1488, needed 20
[debug] HTTP: -> Resumed suspended code
[info] HTTP: Finished intermediate macro file deployprobe.g
[info] Executing nested macro file retractprobe.g on channel HTTP
[debug] ==> Starting code: G30
[debug] HTTP: Suspending code G1 Z0
[debug] HTTP: Suspending code M400
[debug] HTTP: Finished macro file retractprobe.g
[debug] HTTP: Sent G1 Z0, remaining space 1508, needed 28
[debug] HTTP: -> Resumed suspended code
[debug] HTTP: Sent M400, remaining space 1488, needed 20
[debug] HTTP: -> Resumed suspended code
[info] HTTP: Finished macro file retractprobe.g
[debug] Completed G30
[debug] HTTP: ==> Starting code: G30
[debug] Completed G1 Z0
[debug] HTTP: Finished macro file homez.g
[debug] Completed M400
[info] HTTP: Finished macro file homez.g
[debug] Completed G28 Z
[debug] HTTP: ==> Starting code: G28 Z
[debug] IPC#32: Connection closed
```

chrishamm, 2020-01-07:
I might have an idea why this is happening and it could be related to the new M400 code you introduced in combination with the lack of the macro file. Can you try to remove the M400 again and/or create empty deployprobe.g/retractprobe.g?

gtjoseph, 2020-01-08:
I already have empty deployprobe.g and retractprobe.g files.  I did run the test again without the M400 and then added the M400 just for fun. :)  Same result.

chrishamm, 2020-01-08:
Odd, What happens when you delete both macros as well as M400? I ran your macro file at least 10 times on my setup and it worked every single time.

If that doesn't change anything, please share your config.g.

gtjoseph, 2020-01-08:
when you say "it worked" do you mean that your debug log is different than mine and you see the heightmap not loading until the end?

I'm retesting now.

gtjoseph, 2020-01-08:
Yep, it's the presence of either deployprobe.g or retractprobe.g that's causing it.  Even if I have no-op moves in it like `G91 ; G1 X0 Y0 ; G90` heightmap is completed first.

Here's the working debug log.
```
[debug] IPC#29: Got new UNIX connection, checking mode...
[debug] IPC#29: Command processor added
[debug] IPC#29: Received command SimpleCode
[debug] Waiting for execution of G28 Z
[debug] Processing G28 Z
[debug] HTTP: Sent G28 Z, remaining space 1508, needed 28
[info] Executing nested macro file homez.g on channel HTTP
[debug] ==> Starting code: G28 Z
[debug] Waiting for execution of ; homez.g (type 1)
[debug] Processing ; homez.g
[debug] Completed ; homez.g
[debug] Waiting for execution of G30 (type 1)
[debug] Processing G30
[debug] Waiting for execution of G1 Z0 (type 1)
[debug] Processing G1 Z0
[debug] Waiting for execution of G29 S1 P"heightmap.csv" (type 1)
[debug] Processing G29 S1 P"heightmap.csv"
[debug] HTTP: Sent G30, remaining space 1516, needed 20
[debug] HTTP: Sent G1 Z0, remaining space 1488, needed 28
[info] HTTP: Optional macro file deployprobe.g not found
[debug] HTTP: Suspending code G1 Z0
[debug] HTTP: Sent G1 Z0, remaining space 1488, needed 28
[debug] HTTP: -> Resumed suspended code
[info] HTTP: Optional macro file retractprobe.g not found
[debug] HTTP: Suspending code G1 Z0
[debug] HTTP: Sent G1 Z0, remaining space 1488, needed 28
[debug] HTTP: -> Resumed suspended code
[debug] Completed G30
[debug] Completed G1 Z0
[debug] Completed G29 S1 P"heightmap.csv" => Height map loaded from file heightmap.csv
[info] Height map loaded from file heightmap.csv
[debug] HTTP: Finished macro file homez.g
[info] HTTP: Finished macro file homez.g
[debug] Completed G28 Z
[debug] HTTP: ==> Starting code: G28 Z
[debug] IPC#29: Connection closed
```

chrishamm, 2020-01-08:
Ah great, thanks for testing this! My debug log always looked like the last one you sent.

I've prepared a potential fix which I will commit as part of the next DSF update. This fix will make sure flush requests are never resolved until the macro file completion has been actually sent to RRF. I suspect this is why G29 completed before it was supposed to.

chrishamm, 2020-01-15:
This issue should be fixed in DSF 1.2.3.1 but I haven't tested this version in-depth yet. Also, if you give it a try, please look for potential out-of-order execution - it should not occur but if it does, please let me know.

gtjoseph, 2020-01-15:
Yep.  Confirmed fixed.  If I see any others I'll open a new issue.  Thanks!

closed 2020-01-15 by gtjoseph