Duet3D/DuetSoftwareFramework #96: G29 S0 doesn't reports errors when started from macro using /machine/code or CodeConsole

gtjoseph, 2020-01-08:
I have a macro...

special_probe.g
```
M557 X0:100 Y0:350 S5
M558 R0.75   F300   H5.0   T6000   A1        S0
G29 S0
```
A spacing of 5mm exceeds the RRF probe point limit so the M557 should complain.  The G29 should then complain that there's no grid defined.

If I execute the M557 or the G29 from the CodeConsole, the CodeConsole gets the error.
If I execute either from a /machine/code websocket call, I also get the error.

If I execute `M98 P"/macros/special_probe.g"` via the CodeConsole, the errors appear in the DCS log but there's no feedback in CodeConsole.  I do get a message the websocket.

If I execute `M98 P"/macros/special_probe.g"` via the /machine/code, no errors appear in the DCS log and no messages are sent via the websocket.

If I execute`M98 P"/macros/special_probe.g"` from the USB serial console, I get no feedback but I do get the websocket messages.

So, not getting immediate feedback is OK but not getting error messages via the websocket is bad.

chrishamm, 2020-01-09:
Thanks for reporting this. I'll change DCS to append G-code replies from non-system macro codes to the starting code of the macro. However I doubt your report about `/machine/code` is correct, perhaps you just didn't send a proper POST request. When I run a macro file from DCS (which does nothing else than sending M98), I do get messages in the log.

It's expected that M98 from a serial line of the Duet 3 (like USB) does not output any messages generated by DCS. To fix this among a few other inconsistencies, G/M/T codes must be sent from RRF to DCS first before they are processed by DCS and finally a reply is returned. This is planned but low-priority.

gtjoseph, 2020-01-09:
I swear I tested this via machine/code about a dozen times yesterday because I was trying the macro and not understanding why it wasn't actually doing anything.  This morning, I run the macro and I get the errors back in the model via the websocket.

Closing this out.. sorry about the noise.

closed 2020-01-09 by gtjoseph

chrishamm, 2020-01-09:
The fact that M98 doesn't output error messages from macro codes is still a valid bug report so I've fixed it in my local working copy :-)