Duet3D/DuetSoftwareFramework #98: Delays while printing skirt

bjbeare, 2020-01-10:
Slicer: cura
Board: Duet 3 MB6HC (MB6HC)
DSF Version: 1.1.0.5
Firmware: RepRapFirmware for Duet 3 MB6HC v0.6 or 1.0 3.0beta12 (2019-11-02b1)

I notice that while a skirt is being printed, there is a distinct delay at the beginning of each "ring" in the skirt. This seems to be about 100 or 200 ms. Here's what the gcode looks like at the time of the delay.

G1 X139.752 Y141.193 E483.88251
G1 X139.756 Y141.187 E483.88287
G1 X140.226 Y140.464 E483.92589
G1 X140.259 Y140.418 E483.92872
G1 X141.555 Y138.782 E484.03285
G1 X141.596 Y138.736 E484.03592
G0 F3600 X141.882 Y139.015
G1 F1800 X143.276 Y137.588 E484.13545
G1 X143.291 Y137.575 E484.13644
G1 X144.908 Y136.223 E484.24159
G1 X144.928 Y136.208 E484.24284
G1 X148.566 Y133.864 E484.45875
G1 X148.587 Y133.852 E484.45996
G1 X152.639 Y131.897 E484.68441
G1 X152.654 Y131.891 E484.68522

The G0 appears to be the reposition for the new skirt ring. I don't know if the delay is before the G0 or after. Known issue?

wilriker, 2020-01-10:
Hi @bjbeare, please upgrade both DSF and RRF the latest versions and check if this issue still occurs.

That's 1.2.3.0 for DSF and 3.0 (final) for RRF.

bjbeare, 2020-01-10:
Will do once the current print (12 hours) is complete.

bjbeare, 2020-01-12:
Still see the issue although the delay is pretty small.
Board: Duet 3 MB6HC (MB6HC)
DSF Version: 1.2.3.0
Firmware: RepRapFirmware for Duet 3 MB6HC v0.6 or 1.0 3.0 (2020-01-03b3)

Here's a couple of  videos.
https://photos.app.goo.gl/nMuixsEHvqAVRRsF6
https://photos.app.goo.gl/XLzKgwcUfSCVfTPv8

chrishamm, 2020-01-13:
Are you using a heightmap (G29) and if yes, does it make a difference when the heightmap is not in use?

DSF waits 25ms after every SPI transfer but since RRF caches more than 12 G-codes that interval should be sufficient. You can try to decrease this value (SpiPollDelay in /opt/dsf/conf/config.json) and check if it makes a difference.

In the long term, this setting will become obsolete.

bjbeare, 2020-01-21:
I'm no longer seeing the issue. I had to reload my sdcard due to a sdcard failure... so can't really be certain what cleared the issue. I suggest we close this for now and if it recurs, I'll run thru the given suggestions.

closed 2020-01-21 by bjbeare