Duet3D/DuetSoftwareFramework #99: Can't upload files from PrusaSlicer

dc42, 2020-01-12:
See https://forum.duet3d.com/post/126392,

chrishamm, 2020-01-12:
TBH I am not keen on adding a new compatibility layer for the RRF-style rr_ HTTP requests. Perhaps it's a better idea to send a PR to PrusaSlicer to add support the new RESTful HTTP API of DSF.

Kriechi, 2020-04-10:
Also the Cura plugin is affected by the missing `rr_*` HTTP endpoints that RRF2 provided.
https://forum.duet3d.com/post/145068

cdedwards, 2020-06-20:
has this been fixed yet? I am running a duet 3 SBC pi4 with v3.1.1 and prusaslicer 2.2. Attempting to upload via slicer's give 404 error's. The following is from simplify3d and used to work with duet2. 

[D:\3d printer\Duet]curl -v -k --data-binary "c:/temp/Table_Hanger_Bottom.gcode" "http://duet3.local/rr_upload?name=gcodes/test.gcode"
*   Trying fe80::f3e1:629f:8ad:e9a6...
* TCP_NODELAY set
* Connected to duet3.local (fe80::f3e1:629f:8ad:e9a6) port 80 (#0)
> POST /rr_upload?name=gcodes/test.gcode HTTP/1.1
> Host: duet3.local
> User-Agent: curl/7.55.1
> Accept: */*
> Content-Length: 33
> Content-Type: application/x-www-form-urlencoded
>
* upload completely sent off: 33 out of 33 bytes
< HTTP/1.1 404 Not Found
< Date: Sat, 20 Jun 2020 02:14:00 GMT
< Server: Kestrel
< Content-Length: 0
<
* Connection #0 to host duet3.local left intact

[D:\3d printer\Duet]

wilriker, 2020-06-20:
A Pull Request has been submitted to PrusaSlicer to fix this: https://github.com/prusa3d/PrusaSlicer/pull/4345
Unfortunately Prusa did not react to it so far. Maybe it helps if you go over there and vote for it.

LeavOA, 2020-09-20:
I can confirm that sending to Duet from PrusaSlicer 2.2 works (checking the box "start immediately" also works)
It's like magic! thanks for taking care of this (to both the Duet and PrusaSlicer teams)

adammhaile, 2020-09-20:
@chrishamm any idea when the next release will be out that has this new feature?

chrishamm, 2020-09-20:
3.2-b1 provides all the required API calls to implement a rr_ emulation layer but I haven't started working on a plugin yet.

chrishamm, 2021-07-21:
This was implemented as part of a PR for PrusaSlicer some time ago, closing.

closed 2021-07-21 by chrishamm