Duet3D/DuetSoftwareFramework #102: M0 H1 after M25 runs stop.g not cancel.g

gtjoseph, 2020-01-17:
If I run M25 then M0 H1 from the Duet's USB serial console, cancel.g is run as it should.

If I run those commands via the DCS, stop.g is run instead of cancel.g:

```
[debug] Processing G1 X35.657 Y90.842
[debug] Waiting for execution of G1 X35.671 Y90.661
[debug] Processing G1 X35.671 Y90.661
[debug] File: Sent G1 X35.657 Y90.842, remaining space 420, needed 36
[debug] File: Sent G1 X35.671 Y90.661, remaining space 384, needed 36
[debug] IPC#58: Got new UNIX connection, checking mode...
[debug] IPC#58: Command processor added
[debug] IPC#58: Received command SimpleCode
[debug] Waiting for execution of M25
[debug] Processing M25
[debug] HTTP: Sent M25, remaining space 1516, needed 20
[info] Print paused at file position 2611. Reason: User
[debug] Print has been paused at byte 2611
[debug] Cancelled G1 X44.466 Y112.347
[debug] Cancelled G1 X44.286 Y112.322
[debug] Cancelled G1 X43.366 Y112.153
[debug] Cancelled G1 X43.131 Y112.096
[debug] Cancelled G1 X42.208 Y111.808
[debug] Cancelled G1 X41.982 Y111.722
[debug] Cancelled G1 X41.101 Y111.326
[debug] Cancelled G1 X40.060 Y110.714
[debug] Cancelled G1 X40.887 Y111.213
[debug] Cancelled G1 X39.861 Y110.576
[debug] Cancelled G1 X39.101 Y109.980
[debug] Cancelled G1 X38.920 Y109.820
[debug] Cancelled G1 X38.237 Y109.137
[debug] Cancelled G1 X38.076 Y108.956
[debug] Cancelled G1 X37.480 Y108.195
[debug] Cancelled G1 X36.859 Y107.196
[debug] Cancelled G1 X36.772 Y107.036
[debug] Cancelled G1 X37.343 Y107.996
[debug] Cancelled G1 X36.534 Y106.547
[debug] Cancelled G1 X36.462 Y106.380
[debug] Cancelled G1 X36.141 Y105.534
[debug] Cancelled G1 X36.084 Y105.361
[debug] Cancelled G1 X35.938 Y104.837
[debug] Cancelled G1 X35.898 Y104.661
[debug] Cancelled G1 X35.729 Y103.741
[debug] Cancelled G1 X35.645 Y102.596
[debug] Cancelled G1 X35.700 Y103.501
[debug] Cancelled G1 X35.642 Y102.476
[debug] Cancelled G1 X35.642 Y91.384
[debug] Cancelled G1 X35.642 Y91.324
[debug] Cancelled G1 X35.657 Y90.842
[debug] Cancelled G1 X35.671 Y90.661
[info] Executing nested macro file pause.g on channel HTTP
[debug] ==> Starting code: M25
[debug] Waiting for execution of ; pause.g
[debug] Processing ; pause.g
[debug] Waiting for execution of ; called when a print from SD card is paused
[debug] Completed ; pause.g
[debug] Processing ; called when a print from SD card is paused
[debug] Completed ; called when a print from SD card is paused
[debug] Waiting for execution of M83; relative extruder moves
[debug] Processing M83; relative extruder moves
[debug] Waiting for execution of G1 E-5 F600; retract 5mm of filament
[debug] Processing G1 E-5 F600; retract 5mm of filament
[debug] HTTP: Sent M83; relative extruder moves, remaining space 1516, needed 20
[debug] HTTP: Sent G1 E-5 F600; retract 5mm of filament, remaining space 1480, needed 36
[debug] Waiting for execution of G91; relative positioning
[debug] Processing G91; relative positioning
[debug] Waiting for execution of G1 Z5 F600; lift Z by 5mm
[debug] HTTP: Sent G91; relative positioning, remaining space 1460, needed 20
[debug] Processing G1 Z5 F600; lift Z by 5mm
[debug] Waiting for execution of G90
[debug] HTTP: Sent G1 Z5 F600; lift Z by 5mm, remaining space 1424, needed 36
[debug] Processing G90
[debug] HTTP: Sent G90, remaining space 1404, needed 20
[debug] Completed M83; relative extruder moves
[debug] Completed G1 E-5 F600; retract 5mm of filament
[debug] Completed G91; relative positioning
[debug] Completed G1 Z5 F600; lift Z by 5mm
[debug] HTTP: Finished macro file pause.g
[debug] Completed G90
[info] HTTP: Finished macro file pause.g
[debug] Completed M25 =>
Printing paused at X45.2 Y112.4 Z10.0
[debug] HTTP: ==> Starting code: M25 =>
Printing paused at X45.2 Y112.4 Z10.0
[debug] IPC#58: Connection closed
[debug] IPC#59: Got new UNIX connection, checking mode...
[debug] IPC#59: Command processor added
[debug] IPC#59: Received command SimpleCode
[debug] Waiting for execution of M0 H1
[debug] Processing M0 H1
[info] Cancelled print file
[debug] HTTP: Sent M0 H1, remaining space 1508, needed 28
[info] Cancelled printing file 0:/gcodes/sr.gcode, print time was 0h 0m
[info] Executing nested macro file stop.g on channel HTTP
[debug] ==> Starting code: M0 H1
[debug] HTTP: Finished macro file stop.g
[info] HTTP: Finished macro file stop.g
[debug] Completed M0 H1
[debug] HTTP: ==> Starting code: M0 H1
[debug] IPC#59: Connection closed
```

chrishamm, 2020-01-20:
Thanks for reporting this, I'll provide a temporary fix in DSF 1.2.4. It will become obsolete when the Linux interface gets its own task in RRF.

chrishamm, 2020-01-24:
This is fixed using a temporary work-around in DSF 1.2.4.0.

closed 2020-01-24 by chrishamm