Duet3D/DuetSoftwareFramework #104: Various M291 issues

gtjoseph, 2020-01-27:
The "T" parameter isn't in the model.
M292 doesn't clear the model.
Neither M291 nor M292  cause an update on the websocket.

Any "S" mode other than 1 causes "Error: Code is not supported"
If you omit "S" it defaults to 1 and is accepted.
```
      "messageBox" : {
         "axisControls" : [
            0,
            1
         ],
         "message" : "actual message",
         "mode" : 1,
         "seq" : 0,
         "title" : "title"
      },
```
Any other "S" mode causes the error.

If you send the M291 with an S3 parameter from the Duet3 serial console, the command is accepted
but the DSF model reports the mode as `null`

```
      "messageBox" : {
         "axisControls" : [
            0,
            1
         ],
         "message" : "actual message",
         "mode" : null,
         "seq" : 0,
         "title" : "title"
      },
```

chrishamm, 2020-01-27:
See https://github.com/chrishamm/DuetSoftwareFramework#known-issues. Many of these points are also duplicates of 

#50 
#51 
#52 
and at least related to #69 

They will be fixed when support for the new OM has been added and when the SPI protocol has been further enhanced (probably at some point next week). When the required changes are ready, several codes from RRF will be forwarded to DSF and the behaviour of codes interpreted by RRF and DCS should no longer diverge.

gtjoseph, 2020-01-27:
Yeah it's just been a long time without any update and I wanted to create a little "reminder" :)

chrishamm, 2020-04-03:
Blocking message boxes are supported in DSF 1.3.