Duet3D/DuetSoftwareFramework #105: Temperature fluctuations on hot end

bjbeare, 2020-01-28:
I am seeing what looks to be overshoot/undershoot oscillation in my hot end. I am using PID method. The odd thing about this is that if I pause the print and let the temperature stabilize, the fluctuations are gone when I resume.  I am including a picture of this including the pause and subsequent smooth operation.

[image]

Duet Web Control 2.0.4
Board: Duet 3 MB6HC (MB6HC)
DSF Version: 1.2.3.0
Firmware: RepRapFirmware for Duet 3 MB6HC v0.6 or 1.0 3.0 (2020-01-03b3)

[config.g.txt](https://github.com/chrishamm/DuetSoftwareFramework/files/4123430/config.g.txt)

chrishamm, 2020-01-28:
Your DSF/DWC versions are not up-to-date. Please update them.

Regardless of this, I doubt this is related to DSF because RepRapFirmware is in charge for PID control and DSF/DWC only forwards/displays the received temperatures. Provided you've already performed [PID auto calibration](https://duet3d.dozuki.com/Wiki/Tuning_the_heater_temperature_control) and [manually saved](https://duet3d.dozuki.com/Wiki/Tuning_the_heater_temperature_control#Section_Setting_the_model_parameters_manually) the parameters to config.g while making sure config-override.g does not exist, I suggest you open a new topic for this on the [forum](https://forum.duet3d.com/).

If this problem does not occur with the same configuration (and no config-override.g) in standalone operation of the Duet 3, please let me know.

bjbeare, 2020-01-28:
Update is already in progress (takes quite a while).
No override file present. And yes... the pid auto calibration is recent.
I'll re-run auto calibration after the update and re-test.

I'm running this at a bat-out-of-hell speed to see what the limits are. (cura set at 100mm/s). I'm thinking that the part fan is being deflected by the work as the head moves and causing the algorithm to destabilize.

Will report back. Thank you.

bjbeare, 2020-01-28:
Oh... you mentioned stand-along operation of the duet3. I don't really know how to do that. Can you point me to the doc?

chrishamm, 2020-01-28:
Having a fan blow directly at the hotend may cause spikes like those you see. To operate the Duet 3 in standalone mode, see [here](https://duet3d.dozuki.com/Wiki/Getting_Started_With_Duet_3#Section_Running_in_standalone_mode). Basically, it's like running a Duet Ethernet so make sure to add `M552 S1` to your config.g to enable the network interface again. Except for that step it should be sufficient to copy all the files from `/opt/dsf/sd` to another microSD card. You can use FileZilla to do so (connect via `sftp://duet3`).

bjbeare, 2020-01-29:
I reduced the dead time setting to 3.0 from 3.3 and that seems to (at least for now) healed the issue. I'll close this one for now. I'm thinking that the autotune may be a bit off... but autotune is generally regarded IIUC as more of a starting point.

Thanks!

closed 2020-01-29 by bjbeare