Duet3D/DuetSoftwareFramework #109: DCS 1.3.1 crashes when a heater is in "fault" state

gtjoseph, 2020-04-07:
Extruder heater went into fault...
RRF shows:
```
M409 K"heat"                                                                                            
{"key":"heat","flags":"","result":{"bedHeaters":[0,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"chamberHeaters":[-1,-1,-1,-1],"coldExtrudeTemperature"
:160.0,"coldRetractTemperature":90.0,"heaters":[{"active":70.0,"current":24.6,"max":125.0,"min":-273.1,"monitors":[{"action":0,"condition":"
tooHigh","limit":125.0},{"condition":"disabled"},{"condition":"disabled"}],"sensor":0,"standby":55.0,"state":"off"},{"active":225.0,"current
":42.1,"max":280.0,"min":-273.1,"monitors":[{"action":0,"condition":"tooHigh","limit":280.0},{"condition":"disabled"},{"condition":"disabled
"}],"sensor":1,"standby":160.0,"state":"fault"}]}}    
```

DCS crashes and restarts continually with...
```
[fatal] Update task faulted
   System.ArgumentException: Requested value 'fault' was not found.
   at System.Enum.TryParseByName(RuntimeType enumType, String originalValueString, ReadOnlySpan`1 value, Boolean ignoreCase, Boolean throwOnFailure, UInt64& result)
   at System.Enum.TryParseInt32Enum(RuntimeType enumType, String originalValueString, ReadOnlySpan`1 value, Int32 minInclusive, Int32 maxInclusive, Boolean ignoreCase, Boolean throwOnFailure, TypeCode type, Int32& result)
   at System.Enum.TryParse(Type enumType, String value, Boolean ignoreCase, Boolean throwOnFailure, Object& result)
   at System.Enum.Parse(Type enumType, String value, Boolean ignoreCase)
   at DuetAPI.Utility.JsonLowerCaseStringEnumConverter.JsonLowerCaseStringEnumConverterInner`1.Read(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Utility/JsonLowerCaseStringEnumConverter.cs:line 62
   at System.Text.Json.JsonPropertyInfoNullable`2.OnRead(ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.JsonPropertyInfo.Read(JsonTokenType tokenType, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.JsonSerializer.ReadCore(JsonSerializerOptions options, Utf8JsonReader& reader, ReadStack& readStack)
   at System.Text.Json.JsonSerializer.ReadCore(Type returnType, JsonSerializerOptions options, Utf8JsonReader& reader)
   at System.Text.Json.JsonSerializer.Deserialize(String json, Type returnType, JsonSerializerOptions options)
   at DuetAPI.Machine.ModelObject.UpdateFromJson(JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelObject.cs:line 282
   at DuetAPI.Machine.ModelCollectionHelper.UpdateFromJson(IList list, Type itemType, JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelCollection.cs:line 227
   at DuetAPI.Machine.ModelObject.UpdateFromJson(JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelObject.cs:line 264
   at DuetAPI.Machine.ModelObject.UpdateFromJson(JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelObject.cs:line 244
   at DuetAPI.Machine.MachineModel.UpdateFromFirmwareModel(String key, JsonElement jsonElement) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/MachineModel.cs:line 134
   at DuetControlServer.Model.Updater.Run() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/Model/Updater.cs:line 138
[fatal] Update task faulted
   System.ArgumentException: Requested value 'fault' was not found.
   at System.Enum.TryParseByName(RuntimeType enumType, String originalValueString, ReadOnlySpan`1 value, Boolean ignoreCase, Boolean throwOnFailure, UInt64& result)
   at System.Enum.TryParseInt32Enum(RuntimeType enumType, String originalValueString, ReadOnlySpan`1 value, Int32 minInclusive, Int32 maxInclusive, Boolean ignoreCase, Boolean throwOnFailure, TypeCode type, Int32& result)
   at System.Enum.TryParse(Type enumType, String value, Boolean ignoreCase, Boolean throwOnFailure, Object& result)
   at System.Enum.Parse(Type enumType, String value, Boolean ignoreCase)
   at DuetAPI.Utility.JsonLowerCaseStringEnumConverter.JsonLowerCaseStringEnumConverterInner`1.Read(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Utility/JsonLowerCaseStringEnumConverter.cs:line 62
   at System.Text.Json.JsonPropertyInfoNullable`2.OnRead(ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.JsonPropertyInfo.Read(JsonTokenType tokenType, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.JsonSerializer.ReadCore(JsonSerializerOptions options, Utf8JsonReader& reader, ReadStack& readStack)
   at System.Text.Json.JsonSerializer.ReadCore(Type returnType, JsonSerializerOptions options, Utf8JsonReader& reader)
   at System.Text.Json.JsonSerializer.Deserialize(String json, Type returnType, JsonSerializerOptions options)
   at DuetAPI.Machine.ModelObject.UpdateFromJson(JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelObject.cs:line 282
   at DuetAPI.Machine.ModelCollectionHelper.UpdateFromJson(IList list, Type itemType, JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelCollection.cs:line 227
   at DuetAPI.Machine.ModelObject.UpdateFromJson(JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelObject.cs:line 264
   at DuetAPI.Machine.ModelObject.UpdateFromJson(JsonElement jsonElement, Boolean ignoreSbcProperties) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/Base/ModelObject.cs:line 244
   at DuetAPI.Machine.MachineModel.UpdateFromFirmwareModel(String key, JsonElement jsonElement) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetAPI/Machine/MachineModel.cs:line 134
   at DuetControlServer.Model.Updater.Run() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/Model/Updater.cs:line 138
```

chrishamm, 2020-04-07:
This is a known issue as mentioned on the forums - a fix is ready and will be rolled out as part of DSF 1.3.2.

gtjoseph, 2020-04-07:
Oops.  Didn't see it.  Sorry.

chrishamm, 2020-04-14:
Fixed in DSF 1.3.2.

closed 2020-04-14 by chrishamm