Duet3D/DuetSoftwareFramework #110: DCS Pre-1.3.2  Still hangs on macros

gtjoseph, 2020-04-10:
RRF commit https://github.com/dc42/RepRapFirmware/commit/8d38be149db0e18dba6e9f5f6f554cf1c6d87156
was reverted.

```
Duet Control Server v1.3.2
Written by Christian Hammacher for Duet3D
Licensed under the terms of the GNU Public License Version 3
[info] Settings loaded
[info] Environment initialized
[info] Connection to Duet established
[info] IPC socket created at /var/run/dsf/dcs.sock
[debug] IPC#2: Got new UNIX connection, checking mode...
[debug] IPC#2: Subscription processor registered in Patch mode
[debug] IPC#3: Got new UNIX connection, checking mode...
[debug] IPC#3: Command processor added
[debug] IPC#3: Received command ResolvePath
[debug] Requesting update of key boards, seq 0 -> 0
[debug] Requesting update of key directories, seq 0 -> 0
[debug] Requesting update of key fans, seq 0 -> 10
[debug] Requesting update of key heat, seq 0 -> 7
[debug] Requesting update of key inputs, seq 0 -> 0
[debug] Requesting update of key job, seq 0 -> 1
[debug] Requesting update of key move, seq 0 -> 16
[debug] Requesting update of key network, seq 0 -> 2
[debug] Requesting update of key sensors, seq 0 -> 4
[debug] Requesting update of key spindles, seq 0 -> 0
[debug] Requesting update of key state, seq 0 -> 0
[debug] Requesting update of key tools, seq 0 -> 2
[debug] Requesting update of key volumes, seq 0 -> 0
[debug] Updated key boards
[debug] Updated key directories
[debug] Updated key fans
[debug] Updated key heat
[debug] Updated key inputs
[debug] Updated key job
[debug] Updated key move
[debug] Updated key network
[debug] Updated key sensors
[debug] Updated key spindles
[debug] Updated key state
[debug] Updated key tools
[debug] Updated key volumes
[debug] IPC#4: Got new UNIX connection, checking mode...
[debug] IPC#4: Subscription processor registered in Patch mode
[debug] IPC#5: Got new UNIX connection, checking mode...
[debug] IPC#5: Command processor added
[debug] IPC#6: Got new UNIX connection, checking mode...
[debug] IPC#6: Subscription processor registered in Patch mode
[debug] IPC#7: Got new UNIX connection, checking mode...
[debug] IPC#7: Command processor added
[debug] IPC#5: Received command AddUserSession
[debug] IPC#7: Received command AddUserSession
[debug] IPC#8: Got new UNIX connection, checking mode...
[debug] IPC#8: Command processor added
[debug] IPC#8: Received command GetMachineModel
[debug] IPC#8: Connection closed
[debug] IPC#9: Got new UNIX connection, checking mode...
[debug] IPC#9: Subscription processor registered in Patch mode
[debug] IPC#10: Got new UNIX connection, checking mode...
[debug] IPC#10: Command processor added
[debug] IPC#10: Received command AddUserSession
[debug] IPC#11: Got new UNIX connection, checking mode...
[debug] IPC#11: Command processor added
[debug] IPC#11: Received command ResolvePath
[debug] IPC#11: Connection closed
[debug] IPC#12: Got new UNIX connection, checking mode...
[debug] IPC#12: Command processor added
[debug] IPC#12: Received command ResolvePath
[debug] IPC#13: Got new UNIX connection, checking mode...
[debug] IPC#12: Connection closed
[... 83 lines cut ...]
[debug] HTTP: Sent G1 Z5 F3600, remaining space 964, needed 40
[debug] HTTP: Sent G30 P0 X-235 Y-222 Z-99999 ; probe near a leadscrew, remaining space 908, needed 56
[debug] HTTP: Sent G30 P1 X2 Y235 Z-99999 ; probe near a leadscrew, remaining space 852, needed 56
[debug] HTTP: Sent G30 P2 X230 Y-222 Z-99999 S3 ; probe near a leadscrew and calibrate 3 motors, remaining space 788, needed 64
[debug] Completed M561
[debug] HTTP: Suspending code G1 Z5 F3600
[debug] HTTP: Suspending code G30 P0 X-235 Y-222 Z-99999 ; probe near a leadscrew
[debug] HTTP: Suspending code G30 P1 X2 Y235 Z-99999 ; probe near a leadscrew
[debug] HTTP: Suspending code G30 P2 X230 Y-222 Z-99999 S3 ; probe near a leadscrew and calibrate 3 motors
[debug] HTTP: Suspending code G1 Z5 F3600
[debug] HTTP: Suspending code G30 P0 X-235 Y-222 Z-99999 ; probe near a leadscrew
[debug] HTTP: Suspending code G30 P1 X2 Y235 Z-99999 ; probe near a leadscrew
[debug] HTTP: Suspending code G30 P2 X230 Y-222 Z-99999 S3 ; probe near a leadscrew and calibrate 3 motors
[debug] HTTP: Suspending code G1 Z5 F3600
[debug] HTTP: Suspending code G30 P0 X-235 Y-222 Z-99999 ; probe near a leadscrew
[debug] HTTP: Suspending code G30 P1 X2 Y235 Z-99999 ; probe near a leadscrew
[debug] HTTP: Suspending code G30 P2 X230 Y-222 Z-99999 S3 ; probe near a leadscrew and calibrate 3 motors
[info] Starting macro file homeall.g on channel HTTP
[debug] Waiting for execution of ; homeall.g (macro code)
[debug] Waiting for execution of M98 P"homexy.g" (macro code)
[debug] Waiting for execution of M98 P"homez.g" (macro code)
[debug] Completed G29 S2
[debug] Requesting update of key move, seq 16 -> 18
[debug] Updated key move
```

chrishamm, 2020-04-10:
Please share all of the invoked macro files once again and if possible the output of `M122 "DSF"` at the time DCS hangs up.

gtjoseph, 2020-04-10:
[issue-110.zip](https://github.com/chrishamm/DuetSoftwareFramework/files/4463254/issue-110.zip)

I've got a print going on now but I did do an M122 yesterday and the dcs section showed queued commands.  I'll reproduce and get an M122 when this print finishes (about 30 minutes)

gtjoseph, 2020-04-10:
```
M122 DSF
=== Diagnostics ===
RepRapFirmware for Duet 3 MB6HC version 3.01-RC6+3 running on Duet 3 MB6HC v0.6 or 1.0
Board ID: 08DGM-9T66A-G63SJ-6J9F4-3SD6S-1U03B
Used output buffers: 1 of 40 (13 max)
=== RTOS ===
Static ram: 154580
Dynamic ram: 161648 of which 60 recycled
Exception stack ram used: 516
Never used ram: 76412
Tasks: NETWORK(ready,2076) HEAT(blocked,1184) CanReceiv(suspended,3824) CanSender(suspended,1432) CanClock(blocked,1432) TMC(blocked,80) MAIN(running,2612) IDLE(ready,80)
Owned mutexes:
=== Platform ===
Last reset 05:19:20 ago, cause: software
Last software reset at 2020-04-10 09:56, reason: User, spinning module LinuxInterface, available RAM 76628 bytes (slot 2)
Software reset code 0x0010 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x04432000 BFAR 0x00000000 SP 0xffffffff Task 0x4e49414d
Error status: 0
Free file entries: 10
SD card 0 not detected, interface speed: 37.5MBytes/sec
SD card longest block write time: 0.0ms, max retries 0
MCU temperature: min 43.9, current 44.0, max 44.1
Supply voltage: min 25.3, current 25.4, max 25.4, under voltage events: 0, over voltage events: 0, power good: yes
12V rail voltage: min 12.2, current 12.2, max 12.2, under voltage events: 0
Driver 0: standstill, reads 38140, writes 0 timeouts 0, SG min/max not available
Driver 1: standstill, reads 38139, writes 0 timeouts 0, SG min/max not available
Driver 2: standstill, reads 38140, writes 0 timeouts 0, SG min/max not available
Driver 3: standstill, reads 38140, writes 0 timeouts 0, SG min/max not available
Driver 4: standstill, reads 38140, writes 0 timeouts 0, SG min/max not available
Driver 5: standstill, reads 38140, writes 0 timeouts 0, SG min/max not available
Date/time: 2020-04-10 15:16:08
Slowest loop: 4.13ms; fastest: 0.22ms
=== Move ===
Hiccups: 0(0), FreeDm: 375, MinFreeDm: 375, MaxWait: 0ms
Bed compensation in use: none, comp offset 0.000
=== MainDDARing ===
Scheduled moves: 2, completed moves: 2, StepErrors: 0, LaErrors: 0, Underruns: 0, 0  CDDA state: -1
=== AuxDDARing ===
Scheduled moves: 0, completed moves: 0, StepErrors: 0, LaErrors: 0, Underruns: 0, 0  CDDA state: -1
=== Heat ===
Bed heaters = 0 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1, chamberHeaters = -1 -1 -1 -1
=== GCodes ===
Segments left: 0
Movement lock held by null
HTTP* is ready with "M122 D0 S0 F0" in state(s) 0 58 58, running macro
Telnet is idle in state(s) 0
File* is idle in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger* is idle in state(s) 0
Queue* is idle in state(s) 0
LCD is idle in state(s) 0
SBC is idle in state(s) 0
Daemon* is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Slowest loop: 0.47ms; fastest: 0.01ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0) Telnet(0)
HTTP sessions: 0 of 8
- Ethernet -
State: disabled
Error counts: 0 0 0 0 0
Socket states: 0 0 0 0 0 0 0 0
=== CAN ===
Messages sent 31, longest wait 0ms for type 0
=== Linux interface ===
State: 0, failed transfers: 0
Last transfer: 16ms ago
RX/TX seq numbers: 55715/39645
SPI underruns 0, overruns 0
Number of disconnects: 0
Buffer RX/TX: 0/0-0
=== Duet Control Server ===
Duet Control Server v1.3.2.0
HTTP:
Executing macro /opt/dsf/sd/sys/homeall.g, started by M98 P"homeall.g"
> Next stack level
Executing macro /opt/dsf/sd/macros/PrepPrinter.g, started by M98 P"/macros/PrepPrinter.g"
Suspended code: G1 Z5 F3600
Suspended code: G30 P0 X-235 Y-222 Z-99999 ; probe near a leadscrew
Suspended code: G30 P1 X2 Y235 Z-99999 ; probe near a leadscrew
Suspended code: G30 P2 X230 Y-222 Z-99999 S3 ; probe near a leadscrew and calibrate 3 motors
Suspended code: G1 Z5 F3600
Suspended code: G30 P0 X-235 Y-222 Z-99999 ; probe near a leadscrew
Suspended code: G30 P1 X2 Y235 Z-99999 ; probe near a leadscrew
Suspended code: G30 P2 X230 Y-222 Z-99999 S3 ; probe near a leadscrew and calibrate 3 motors
Suspended code: G1 Z5 F3600
Suspended code: G30 P0 X-235 Y-222 Z-99999 ; probe near a leadscrew
Suspended code: G30 P1 X2 Y235 Z-99999 ; probe near a leadscrew
Suspended code: G30 P2 X230 Y-222 Z-99999 S3 ; probe near a leadscrew and calibrate 3 motors
Number of flush requests: 1
> Next stack level
Code buffer space: 4096
Configured SPI speed: 8000000 Hz
Full transfers per second: 38.03

```

chrishamm, 2020-04-10:
Thanks, I can reproduce the hang and I am looking into it.

chrishamm, 2020-04-11:
It should be fixed in the latest DSF commit. If you are using M99 or blocking message boxes with the option to cancel, I suggest you rebuild RRF with my latest commit, too.

gtjoseph, 2020-04-11:
Confirmed, getting messages and macros running!   One request if possible...  If I add M117 progress messages to the PrepPrinter macro, I get them as they are reached which is good.  The PrepPrinter macro does 3 bed though and the 3 results messages all come together when the PrepPrinter macro finishes.  Any way to get them as they happen?

closed 2020-04-11 by gtjoseph

chrishamm, 2020-04-11:
That's great, thanks once more for testing! To see messages when a certain step is performed I usually use M118 S, but perhaps it's worth adding a new setting to output macro code messages right when they are emitted.