Duet3D/DuetSoftwareFramework #111: Printing a file that has no recognized slicer causes fatal exception in DCS

gloomyandy, 2020-04-10:
Hi,
using 1.3.1 of DSF an attempt to print a file that DSF has not been able to assign a slicer name to will cause a fatal exception in DCS. The following test gcode file:
```
M564 H0
M564
G1 X0 Y0
G1 X200 Y200
G1 X0 Y0
G1 X200 Y200
G1 X0 Y0
G1 X200 Y200
```

which does not contain any slicer identification causes DCS to abort with this exception:
```
[debug] Processing G1 X0 Y0
[fatal] Abnormal program termination
[fatal] SPI task faulted
   System.ArgumentNullException: String reference not set to an instance of a String. (Parameter 's')
   at System.Text.Encoding.GetBytes(String s)
   at DuetControlServer.SPI.Serialization.Writer.WritePrintStarted(Span`1 to, ParsedFileInfo info) in /home/pi/DuetSoftwareFramework/src/DuetControlServer/SPI/Serialization/Writer.cs:line 416
   at DuetControlServer.SPI.DataTransfer.WritePrintStarted(ParsedFileInfo info) in /home/pi/DuetSoftwareFramework/src/DuetControlServer/SPI/DataTransfer.cs:line 636
   at DuetControlServer.SPI.Interface.Run() in /home/pi/DuetSoftwareFramework/src/DuetControlServer/SPI/Interface.cs:line 588
[debug] Processing G1 X200 Y200
[debug] Processing G1 X0 Y0
[debug] Processing G1 X200 Y200
```
I looks like in method WritePrintStarted info.GeneratedBy is set to Null in this case. Not sure if the best fix is to ensure that it is always valid or to add tests for the Null condition.

NOTE: I'm looking after the LPC port of RRF these days and so the above error was found while testing that port, not on a Duet board. However I don't think that this problem is specific to the LPC version.

chrishamm, 2020-04-10:
Thanks, I'll fix it in DSF 1.3.2.

chrishamm, 2020-04-14:
Fixed in DSF 1.3.2.

closed 2020-04-14 by chrishamm